FabMo/FabMo-Engine #263: Create new macro

Ryan-Patterson, 2015-11-25:
I am unable to create a new macro.  I do not have any macros in the list and when I click create macro nothing happens.  I tried from and Android phone and a PC laptop.

ryansturmer, 2015-11-25:
Right click inside the macro manager and choose "Reload Frame" or "This Frame->Reload" and try again, just in case you are getting an old version of the macro manager.  Do this from the laptop.

ryansturmer, 2015-11-25:
Also, what browser, on the laptop?

Ryan-Patterson, 2015-11-25:
The Browser is Chrome this is the first time this laptop was connected to Fabmo.   I am having an issue getting my newest laptop to connect.

Ryan-Patterson, 2015-11-25:
I tried the reload frame.  Also tried Chrome, FireFox and IE.  IE did not display correctly at all.

ryansturmer, 2015-11-25:
IE is not supported.  I am looking into this.

ryansturmer, 2015-11-25:
Wait, you've got one laptop that will connect and a new one that doesn't?

Ryan-Patterson, 2015-11-25:
Yes I started a new issue on the connection.  

I have now tried creating a Macro on a Windows10 tablet and it will not create a new one.  It shows not data available in table.

Ryan-Patterson, 2015-11-25:
Not sure if this helps but if I go to this page 192.168.42.1:9876/approot/macro_manager.fma/index.html# and click new macro there is a little box that displays briefly it reads something about a call to update macro.

ryansturmer, 2015-11-27:
Ryan, if you're wanting to provide diagnostic information - visit the macro manager in chrome, and open the developers console (Ctrl+Shift+J) and click "create new macro" and post any messsages that are printed to the console

Ryan-Patterson, 2015-11-29:
I do not get any messages.  If I refresh the page I get "Websocket connected"

ryansturmer, 2015-11-30:
If you get the chance to bring your card into the shop - Might be best that I observe the issue myself!

commit 281c117b (2015-11-30): Fix #263
closed 2015-11-30 by ryansturmer