FabMo/FabMo-Engine #264: Connect Win10 to WIFI (AP Mode)

Ryan-Patterson, 2015-11-25:
My Windows10 computer will not connect to the FabMo WIFI.  When I try to connect it does not pop up to enter.  It will say it is checking connection for about 30 seconds then report is was not abe to connect.

Ryan-Patterson, 2015-11-25:
My Windows10 tablet is able to connect to the Fabmo WIFI.  I have also tried to connect FabMo to my home WIFI and it will switch.  I double click the WIFI connection enter the passcode and click connect.

ryansturmer, 2015-11-25:
Ryan - what is the output of `ipconfig` after you have connected to AP mode from your win10 laptop but it reports no connection?

Ryan-Patterson, 2015-11-25:
The Win10 laptop will not connect.
When I click to connect to displays "Checking Network requirements" then laptop will connect back to my home WIFI.  When trying to connect to FabMo it does not ask me for a password.

ryansturmer, 2015-11-25:
@tedh-shopbot @gordonab  - have you guys connected from win 10?

gordonab, 2015-11-25:
I haven't yet. -g

Ryan-Patterson, 2015-11-25:
Stupid mistake on my part.  The laptop that would not connect was connected to Fabmo WFFI then the card was updated with a new WIFI password the laptop had the old one stored.  Once I cleared the password it connects.  Sorry

Still have the issue with creating a new macro and getting fabmo connected to home wifi.

closed 2015-11-26 by Ryan-Patterson