FabMo/FabMo-Engine #265: Files don't always update

wlyoung, 2015-11-30:
This could easily be operator error, but I save files in a dropbox folder to run with FabMo. Occasionally I'll run an sbp file and realize I need to make a small change. I make the change in vcarve and then replace the file in Dropbox.

In FabMo I go through the Submit job process and select that re-jiggered file to cut, but sometimes it doesn't seem to actually get to new file and will rerun the original. To be sure that it's running the modified file I find I need to rename the file.

ryansturmer, 2015-11-30:
When you submit the new file and run it and it appears not to run, if you download or edit that file from the history - does it show the edited file, or the old one?  Are you certain that your dropbox is fully synced before re-running?

wlyoung, 2015-11-30:
Like I said it could easily be operator error. If I open it in the editor
the old values are in there but I can't make it happen on purpose. I'm
thinking that I submitted the modified version as a new job but then just
clicked the run arrow, assuming that it had been replaced? Is that cached
somewhere?

I know this isn't the first time I've whined about this, but I still find
the system for running a file a bit cumbersome. Sometimes the file shows up
to run immediately after I submit it, but at others the last file that was
run is teh one that has the run arrow, and I have to open pending files to
find it. It's just too many steps in my whiny opinion.

wlyoung, 2015-11-30:
And it's certainly possible that it's a dropbox failure, but when I rename
and go through the same process it's always there. Does take a second or
two to populate in the Fabmo file browser though...I may have jumped the
gun and it hadn't refreshed completely when I clicked it?

ryansturmer, 2015-11-30:
Bill your opinion is not whiny - remember this is a draft version of the interface, we are working on streamlining things.  Thank you for your feedback - please know that it is being incorporated into the design of subsequent versions.

Remember that this is a queue, just like a printer. If you submit five files before running any of them, the first one you submitted runs first, followed by the second, etc.  If you go browse for a file, and send it to the tool, and then change your mind and go pick another file, the first file is still there, waiting in the queue to run.  The most recent file selected isn't promote to "next" - next is next, until you either decide it's not or you run it.  If you see that big play button, and you go browse for a file and then click it - you're not running the file you just browsed for.

I don't suspect your issue is one of refresh time.  That shouldn't happen.  If the play button is there, that means there's a job in the queue, ready to run.  The only way for a job to get there is for you to send it (via browsing for it, or using an app) or for a job to be re-run from the history.  Now, there may of course be a bug  - and old jobs are being cached and resubmitted somehow - that's what we're angling to find out.  I'm just trying to get the error reproduced so we can fix it.

Ryan-Patterson, 2015-11-30:
I am seeing the same thing, edit a file and it is using the older file.  On my android tablet the submit job button no longer works.

ryansturmer, 2015-12-01:
@Ryan-Patterson Re: the older file issue - how are you submitting the job, and on what platform?  (android, windows, etc) are you using the "Submit job" button, or dragging into the job manager, or clicking the link when there's an empty queue, or using SB4?

Ryan-Patterson, 2015-12-01:
It is android selecting a file from DropBox by clicking Submit job button.  The files has been updated.

closed 2015-12-08 by ryansturmer