FabMo/FabMo-Engine #266: stutter circles

Ryan-Patterson, 2015-11-30:
Circles are not smooth.  Attached is the file
[FabMo_Test.txt](https://github.com/FabMo/FabMo-Engine/files/47955/FabMo_Test.txt)

ryansturmer, 2015-12-01:
Ryan - is it a non-smoothness of motion, or of the actual cut?  Can you post an image of the final cut so we can see how it looks?

Ryan-Patterson, 2015-12-01:
I did not cut the part just not smooth motion.  It is a start and stop as it moves.  I did try posting out as line segments and it moved about the same (Not smooth) but slower.

ryansturmer, 2015-12-01:
What are your junction acceleration and chordal tolerance settings (in the configuration page)

Ryan-Patterson, 2015-12-01:
I think it is .001  I did try at .01 and .0001 I did not see any change.

ryansturmer, 2015-12-04:
Closing this as a junction acceleration/deviation issue - if we can't arrive at satisfactory settings, we can reopen.

closed 2015-12-04 by ryansturmer