FabMo/FabMo-Engine #267: Setting changes in a file

Ryan-Patterson, 2015-12-01:
Can it be made for setting commands used within a file to revert back to the way it was before the file was run?  The reason I ask is if I use the VI command to swap an axis from within a file and stop the file before it ends it will not revert back.

ryansturmer, 2015-12-01:
Settings commands are no longer to be supported in files, they will be in the configuration view only.

Ryan-Patterson, 2015-12-01:
So I we will not be able to do an axis swap in a file or change the speed?

ryansturmer, 2015-12-01:
Speed changes of course will be allowed, but AFAIK you can't do an axis swap, or change unit values, these will only be supported through the configuration interface.  That's the plan we have now.

Ryan-Patterson, 2015-12-01:
This would be an issue for a dual Z setup.

ryansturmer, 2015-12-01:
This will be a topic of discussion as we move to larger tools, but since there are no dual-Z handibots, for now, configuration for unit, axis setup, etc won't be done in OpenSBP files.  We're going to put together a command listing for allowed commands, since FabMo only supports a subset of the full OpenSBP standard.

I'm leaving this open though, as a topic for discussion.

gordonab, 2015-12-01:
Right now the VI command is supported in files. At least, until we turn off
these commands. The unit values [VU] should still be working, but is going
away. [VA],[VI],[VC],[VD],[VL],[VR]&[VS] all work, but will be on the list
of potential drops.

ryansturmer, 2015-12-01:
Since we're dropping it, I'm not looking into making the VI axis swaps go back at the end of files - if we decide to reintroduce them later - that can be part of the discussion

Ryan-Patterson, 2015-12-01:
Why drop them if they are working?  Are you going to have something else in place to be able to use a second Z without having to use a custom post?

Some of the commands that Gordon has listed looks like they would be in a file like [VA] for setting zeroing of table base and the [VS] for setting speed on a 5 axis.

ryansturmer, 2015-12-01:
There are practical considerations to using the commands with the system the way it is now that we want to think about more before we implement them - for instance the issue you raise about what happens when command persistence is desired only within a file - or what impact changes to configuration of the tool in the OpenSBP world has on how G-code files run - or how the system behaves differently between "in-file" commands and "immediate" commands - it's a larger discussion that I think we've agreed we want to pay due dilligence to and get right, rather than charge ahead and mimick the way the system works now, which may not be the best thing to do, given the new motion platform and the network-based model for how files are loaded and run.  Don't worry - we're not categorically removing features, we're just being more conservative with what is supported *today* given that we're pushing up to a big release.

Ryan-Patterson, 2015-12-01:
Cool.  I know most of what I am looking for HandiBot users will not need.  I do agree a better way may exist from what we are doing now.  Just not wanting to lose function or capability.

ryansturmer, 2015-12-01:
It's our intent that the new system be **more** capable not less, but we're having to postpone some of the more advanced features to get working systems on the ground now

ryansturmer, 2016-01-07:
I feel that we have a handle on this, the design has been done, and much of OpenSBP has been adjusted to accommodate it, so I'm going to close this.  We can reopen if we find as we move to bigger tools that we don't have the capability we want.

closed 2016-01-07 by ryansturmer