FabMo/FabMo-Engine #269: Job Manager closing stop buttons while file running ...

tedh-shopbot, 2015-12-01:
When I run the s_logo.sbp file, immediately after the pause to spin up the router, the job manager is reverting to the regular DRO and removing the stop/pause/progress system. I.e. no way to stop the file. At first I thought this only happened from Sb4 call ... but seems universal.

ryansturmer, 2015-12-01:
Can you attach the file here, Ted?

Ryan-Patterson, 2015-12-03:
Just had this happen.  I was trying the vinyl file from other issue and I got the circle waiting for Fabmo I refreshed the browser the file was running but it did not display the stop or progress bar.  When moved the spinner the file stopped and it moved based on what I was doing with the spinner.

ryansturmer, 2015-12-09:
This is sort of part of a larger issue.  This file causes the tool to choke entirely:

https://gist.github.com/ryansturmer/3d870fb5ba660b45f146

ryansturmer, 2016-01-07:
No longer!

closed 2016-01-07 by ryansturmer