FabMo/FabMo-Engine #273: Long load time and incomplete cut

Ryan-Patterson, 2015-12-03:
I submit the attached file as a job and click to run.  The spindle starts within a couple seconds then movement does not start until after 30 plus seconds.  Then on my card at home the motion will stop a few lines into the file.  Tried same file on Desktop in back the startup is the same but it does rum to completion.  Starting file from Android phone on both controllers.
[Vinyl.txt](https://github.com/FabMo/FabMo-Engine/files/50996/Vinyl.txt)

gordonab, 2015-12-03:
Runs to completion on the fabmo office DT. It takes about the same amount
of time to load, loading 14722 lines. Did you change the name to Vinyl.sbp
to run it?

Ryan-Patterson, 2015-12-03:
Yes I changed the name to txt to be able to uplaod the file.

ryansturmer, 2015-12-03:
Ryan, can you post the contents of /opt/fabmo/config/engine.json on your tool?

ryansturmer, 2015-12-03:
If you have the card with you I can look at it this afternoon, also

Ryan-Patterson, 2015-12-03:
I have the card at home.

ryansturmer, 2015-12-03:
The contents of that file, and the contents of the debug log after the file fails will be instructive.

ryansturmer, 2015-12-03:
Actually, ryan, if you go to your tool url/config and paste the result here - that will save you from having to go fish out that file.

Ryan-Patterson, 2015-12-03:
Thought it might be my home network so I tried in AP mode and same thing occurs.
I will bring my card in tomorrow.

ryansturmer, 2015-12-04:
Long load time is fixed in the latest version.  (Print statements in the OpenSBP runtime really slow things down due to writing the journal) - stop in the middle is unexpected though as it doesn't happen on other tools - is it possible that you hit an input switch, or got a spurious input trigger?

-R

ryansturmer, 2015-12-04:
Updated and retested, seems to be fixed - I think this was a corruption issue that was causing the crash, but will keep an eye on it.

closed 2015-12-04 by ryansturmer