FabMo/FabMo-Engine #274: Hide the red panel when doing non-movement stuff?

wlyoung, 2015-12-04:
When running Fabmo on a phone, the red panel takes up a lot of real estate. If it's extended and you go to the queue or rmacro menu or other screen that won;'t cause motion without you taking another action, could the red panel auto-hide? You have to hide it anyway to be able to do anything else..it would be nice if it just slid away on its own

ryansturmer, 2016-04-22:
Thoughts on this:

1. Bringing the DRO out at the start of a job is not essential: it brings focus to the positional display that is not needed or wanted
2. The prevailing thinking is that the DRO should only ever show or hide when prompted by the user, or prompted by an app.

wholetthedogsout, 2016-04-25:
So currently we do not actually bring out the DRO on job start.  Do we want to force hide the DRO on job start on mobile as Bill ? @tedh-shopbot @wlyoung @ryansturmer @jlucidar

ryansturmer, 2016-04-26:
My opinion is that we never want to open/close the DRO unless the user asks explicitly by clicking the hamburger.

closed 2016-04-26 by wholetthedogsout