FabMo/FabMo-Engine #283: G2 or G3 wrong parameters crash the engine then issues when reconnection

Alex-Canales, 2015-12-10:
I tried this command:

    G0 X0 Y0 Z0
    G2 I0 J0 F200
    G0 X2 Y2 Z2

I and J parameters are wrong. After trying this code, it was not possible to connect to the tool (so I suppose the engine crashed). I restarted the machine.

I can now be connected and send commands by using the hand-wheel. It displays that the position is changing but nothing is moving physically.

Or I am connected to an other (thanks Ryan for the explanation by the way) machine or I receive wrong data.

ryansturmer, 2015-12-11:
Alex!  I completely steered you wrong with "an other" versus "another" - this is not a software issue, but it is one of syntax (or rather of semantics) so I'll clarify here:

another only works as an adjective when the qualified noun is singular - it can mean additional *or* alternate.  It depends on context.

other only works as an adjective when the qualified noun is plural - it can also mean additional *or* alternate.  It depends on context.

So "an other" is incorrect, because the indefinite article "an" is singular, and "other" is an adjective for qualifying plural nouns.

Example:
We should write another function to do this.
We should write other functions to do this.

Of course there are exceptions here, but this is the most sensible "rule of thumb" explanation I can come up with.

ryansturmer, 2015-12-11:
Don't know why the engine is crashing though.

gordonab, 2015-12-11:
ab3fd34: adds checks in all circle and arc commands for zero or negative diameter.

closed 2016-01-21 by ryansturmer