FabMo/FabMo-Engine #286: IF statement issue with Macros ???

tedh-shopbot, 2015-12-12:
Still struggling this weekend with getting macros to run ... and to run in files.
But here's one related to just running the macro manually. Forgive all the commented lines in the samples ... but I was just trying to pare this down to nothingness to isolate issues.

>>>>>THIS ONE RUNS--------------------------------------

'!FABMO!name:BEND
'!FABMO!description:Bender: BEND Function (w/duck & comp)
'Bender: MAKE A BEND

'Position Bend Pin and Bend depending on whether CW or CC bend
'IF &next_bend > 0 THEN GOSUB CW_BEND
'IF &next_bend < 0 THEN GOSUB CC_BEND

'END

'Subroutines
    CW_BEND:
        C#,33
'        MX, (&startBend_offset / 2) + (&next_bend)
         MX, 50
'    RETURN

'    CC_BEND:
'        C#,34
'        MX, (-1 * &startBend_offset / 2) + &next_bend
'    RETURN

>>>>>THIS ONE LOCKS UP WITH PAUSE/STOP SHOWING--------------------------------------
(usually requiring a reboot) ... sometimes it will restart and can then be stopped; but stop itself never works when locked.

'!FABMO!name:BEND
'!FABMO!description:Bender: BEND Function (w/duck & comp)
'Bender: MAKE A BEND

'Position Bend Pin and Bend depending on whether CW or CC bend
IF &next_bend > 0 THEN GOSUB CW_BEND
'IF &next_bend < 0 THEN GOSUB CC_BEND

END

'Subroutines
    CW_BEND:
        C#,33
'        MX, (&startBend_offset / 2) + (&next_bend)
         MX, 50
    RETURN

'    CC_BEND:
'        C#,34
'        MX, (-1 * &startBend_offset / 2) + &next_bend
'    RETURN

>>>>>THIS ONE RUNS--------------------------------------

'!FABMO!name:BEND
'!FABMO!description:Bender: BEND Function (w/duck & comp)
'Bender: MAKE A BEND

'Position Bend Pin and Bend depending on whether CW or CC bend
'IF &next_bend > 0 THEN GOSUB CW_BEND
'IF &next_bend < 0 THEN GOSUB CC_BEND

'END

'Subroutines
    CW_BEND:
        C#,33
'        MX, (&startBend_offset / 2) + (&next_bend)
         MX, 50
'    RETURN

'    CC_BEND:
'        C#,34
'        MX, (-1 * &startBend_offset / 2) + &next_bend
'    RETURN

>>>>>THIS ONE DOES NOTHING WHEN YOU CLICK MACRO START BUTTON---------------
(other macros still work) ... sometimes after running this, the entire contents of the macro disappears from FabMo, though the name of the macro does not.

'!FABMO!name:BEND
'!FABMO!description:Bender: BEND Function (w/duck & comp)
'Bender: MAKE A BEND

'Position Bend Pin and Bend depending on whether CW or CC bend
IF &next_bend > 0 THEN GOSUB CW_BEND
*or optionally with this line here too: MY, 10
END  
'IF &next_bend < 0 THEN GOSUB CC_BEND

'END

'Subroutines
    CW_BEND:
        C#,33
'        MX, (&startBend_offset / 2) + (&next_bend)
         MX, 50
    RETURN

'    CC_BEND:
'        C#,34
'        MX, (-1 * &startBend_offset / 2) + &next_bend
'    RETURN

ryansturmer, 2015-12-13:
Ted can you furnish the contents of macro 33 and 34?

tedh-shopbot, 2015-12-13:
See it you have a "FabMo-Apps from_07_15" folder in your shared dropbox. There is a macro folder in there that is a basic copy of all the macros I have been playing with. At this instant, some of the ones on my DIWire-FabMo may be commented and modified to debug. But those are the basic macros ...

ryansturmer, 2015-12-21:
Fixed in 4260173

closed 2015-12-21 by ryansturmer