FabMo/FabMo-Engine #290: Math Issue in SBP language ...

tedh-shopbot, 2015-12-20:
MX, (((-1 * &startBend_offset) / 2) + &this_bend)   'goes to a different place than
        MX, ((-1 * &startBend_offset / 2) + &this_bend)

not such a big deal itself, but maybe symptomatic ...?

ryansturmer, 2015-12-20:
What two places do they go?

ryansturmer, 2015-12-20:
This is definitely a big deal, language wise, but may be easy to fix

ryansturmer, 2015-12-20:
Looks like the first one goes to the right place, and the second one doesn't.

ryansturmer, 2015-12-20:
`MX,[{"left":{"left":{"left":-1,"right":"&startBend_offset","op":"*"},"right":2,"op":"/"},"right":"&this_bend","op":"+"}]` (correct, with parens)

versus

`MX,[{"left":{"left":-1,"right":"&startBend_offset","op":"*"},"right":"&this_bend","op":"+"}]` (incorrect)

ryansturmer, 2015-12-21:
This is a problem with multiple operators with the same precedent being used in the same expression.  `1+2+3` doesn't work either.

tedh-shopbot, 2015-12-21:
I thinking it was confusing for me to follow-up on this in slack ...

ryansturmer, 2015-12-21:
I see your messages now!  I have my head around this, and am working on the fix.  It's a straightforward parse tree issue.

commit eaaba20b (2015-12-21): Fix #290
closed 2015-12-21 by ryansturmer