FabMo/FabMo-Engine #291: Rotation/Scale/etc values being set, but not be read or cleared

tedh-shopbot, 2015-12-20:


gordonab, 2015-12-20:
Since the transform values are only stored in the opensbp.sbp file they are
read with the get or getMany functions. Likewise, to clear them, set or
setMany are used to set them to disabled so that they aren't performed.

Right now there aren't any default values for these transforms to be reset
to.

These aren't modal at this time and there aren't any corresponding OpenSBP
commands. The rotation of rectangles in the CR command is modal to file
operation, being isolated to that function. Any Dr with a rotation would be
transformed again by the overall transforms.

How do you expect these to work?

-Gordon

ryansturmer, 2016-01-08:
Fixed the setting/getting of these params.

closed 2016-01-08 by ryansturmer