FabMo/FabMo-Engine #295: Job Manager cleanup of up-next card

ryansturmer, 2016-01-04:
Still acts kind of funny - at first load of the page, it often has content that seems to dismiss immediately (should start blank and then get filled in)

On top of that, there is occasional weirdness:

[image]

wholetthedogsout, 2016-01-04:
Working on this now, there was a lot of messy logic to get it to a functional state but doing a lot of house cleaning now

wholetthedogsout, 2016-01-04:
I think I previously fixed the overlapping text. Do you know what conditions led to this?

ryansturmer, 2016-01-21:
I haven't seen this in a while, I think it's resolved.

closed 2016-01-21 by ryansturmer