FabMo/FabMo-Engine #299: Gcode circle goes into infinite loop

gordonab, 2016-01-07:
The following code will put G2 into an infinite loop. It will run the last circle and Z pull-up indefinitely. I've only let it run as long as 10 mins. This is what is being sent to G2:

N0G0Z0.25000F240.00000
N0G0X0.37500Y0.43750
N0G1Z0.00000F240.00000
N0G2X0.37500Y0.43750I0.00000J-0.06250Z-0.27000F240.00000
N0G0Z0.00000
N1G1Z0.25000F120.00000
N2G0X0.37500Y6.44250
N2G2X0.37500Y6.44250I0.00000J-0.06250Z-0.27000F240.00000
N2G0Z0.25000

ryansturmer, 2016-01-08:
See #300 for the G2 configuration under which this behavior happened.

gordonab, 2016-01-28:
This is the latest code that causes the infinite circle issue when the engine is running on my PC. Checking the config file and will run on an Edison as well.

N0 G0Z0.25000F360.00000
N0 G0X0.37500Y0.43750
N0 G1Z0.00000F360.00000
N0 G2X0.37500Y0.43750I0.00000J-0.06250Z-0.27000F360.00000
N0 G2X0.37500Y0.43750I0.00000J-0.06250F360.00000
N0 G0Z0.00000
N1 G1Z0.25000F120.00000
N2 G0X0.37500Y6.44250
N2 G0Z0.00000
N2 G2X0.37500Y6.44250I0.00000J-0.06250Z-0.27000F360.00000
N2 G2X0.37500Y6.44250I0.00000J-0.06250F360.00000
N2 G0Z0.25000
N4 G0X-0.06250Y0.37500
N4 G0Z0.00000
N4 G1Z-0.27000F120.00000
N4 G3X0.81250Y0.37500I0.43750J0.00000F360.00000
N5 G0X0.81250Y6.38000
N5 G3X-0.06250Y6.38000I-0.43750J0.00000F360.00000
N6 G1Y0.37500F360.00000
N7 G1Z0.25000F120.00000

ryansturmer, 2016-02-02:
Closing as windows-only node-serialport issue, and underlying parse issue has been reported to synthetos.

closed 2016-02-02 by ryansturmer