FabMo/FabMo-Engine #300: Circle code panics G2

gordonab, 2016-01-07:
This series of moves and circles will panic the G2 controller ever time. If the circles are run individually they work fine.

Code sent to G2:

N0G0Z0.25000F240.00000
N0G0X0.37500Y0.43750
N0G1Z0.00000F240.00000
N0G2X0.37500Y0.43750I0.00000J-0.06250Z-0.27000F240.00000
N0G2X0.37500Y0.43750I0.00000J-0.06250F240.00000
N0G0Z0.00000
N1G1Z0.25000F120.00000
N2G0X0.37500Y6.44250
N2G2X0.37500Y6.44250I0.00000J-0.06250Z-0.27000F240.00000
N2G0Z0.25000
M30

M30 [g2]

Log output:

g2: <-C--1452174763304---- {"sr":{"posy":2.899,"vel":337.87}} [g2]
g2: <-C--1452174763556---- {"sr":{"posy":4.383,"vel":359.98}} [g2]
g2: <-C--1452174763809---- {"sr":{"posy":5.894,"vel":360.00}} [g2]
g2: <-C--1452174763917---- {"er":{"fb":83.09,"st":31,"msg":"Move time is NAN - prep isnan"}} [g2]
g2: <-C--1452174763918---- {"r":{},"f":[1,206,13]} [g2]
error: 206,STAT_COMMAND_REJECTED_BY_PANIC,command was not processed because machine is paniced [machine]
g2: <-C--1452174763920---- {"r":{},"f":[1,206,23]} [g2]
error: 206,STAT_COMMAND_REJECTED_BY_PANIC,command was not processed because machine is paniced [machine]
g2: <-C--1452174763922---- {"r":{},"f":[1,206,21]} [g2]
error: 206,STAT_COMMAND_REJECTED_BY_PANIC,command was not processed because machine is paniced [machine]
g2: <-C--1452174763923---- {"r":{},"f":[1,206,27]} [g2]
error: 206,STAT_COMMAND_REJECTED_BY_PANIC,command was not processed because machine is paniced [machine]
g2: <-C--1452174763925---- {"r":{},"f":[1,206,23]} [g2]
error: 206,STAT_COMMAND_REJECTED_BY_PANIC,command was not processed because machine is paniced [machine]
g2: <-C--1452174763926---- {"r":{},"f":[1,206,21]} [g2]
error: 206,STAT_COMMAND_REJECTED_BY_PANIC,command was not processed because machine is paniced [machine]
g2: <-C--1452174763926---- {"r":{},"f":[1,206,23]} [g2]
error: 206,STAT_COMMAND_REJECTED_BY_PANIC,command was not processed because machine is paniced [machine]
g2: <-C--1452174763943---- {"r":{},"f":[1,206,57]} [g2]
error: 206,STAT_COMMAND_REJECTED_BY_PANIC,command was not processed because machine is paniced [machine]
g2: <-C--1452174763944---- {"r":{},"f":[1,206,48]} [g2]
error: 206,STAT_COMMAND_REJECTED_BY_PANIC,command was not processed because machine is paniced [machine]
g2: <-C--1452174763945---- {"r":{},"f":[1,206,13]} [g2]
error: 206,STAT_COMMAND_REJECTED_BY_PANIC,command was not processed because machine is paniced [machine]
g2: <-C--1452174763946---- {"r":{},"f":[1,206,23]} [g2]
error: 206,STAT_COMMAND_REJECTED_BY_PANIC,command was not processed because machine is paniced [machine]
g2: <-C--1452174763947---- {"r":{},"f":[1,206,21]} [g2]
error: 206,STAT_COMMAND_REJECTED_BY_PANIC,command was not processed because machine is paniced [machine]
g2: <-C--1452174763947---- {"r":{},"f":[1,206,57]} [g2]
error: 206,STAT_COMMAND_REJECTED_BY_PANIC,command was not processed because machine is paniced [machine]
g2: <-C--1452174763948---- {"r":{},"f":[1,206,13]} [g2]
error: 206,STAT_COMMAND_REJECTED_BY_PANIC,command was not processed because machine is paniced [machine]
g2: <-C--1452174763949---- {"r":{},"f":[1,206,4]} [g2]
error: 206,STAT_COMMAND_REJECTED_BY_PANIC,command was not processed because machine is paniced [machine]
g2: <-C--1452174763950---- {"r":{},"f":[1,206,4]} [g2]
error: 206,STAT_COMMAND_REJECTED_BY_PANIC,command was not processed because machine is paniced [machine]
g2: <-C--1452174763955---- {"er":{"fb":83.09,"st":94,"msg":"Planner assertion failure - zero length line in calculate tr
apezoid"}} [g2]
g2: <-C--1452174763956---- {"er":{"fb":83.09,"st":94,"msg":"Planner assertion failure - min time move in mp_plan_block_l
ist"}} [g2]
g2: <-C--1452174763957---- {"er":{"fb":83.09,"st":94,"msg":"Planner assertion failure - buffer empty3 in mp_plan_block_l
ist"}} [g2]
g2: <-C--1452174764065---- {"sr":{"posx":0.000,"posy":0.000,"posz":0.000,"vel":0.00,"stat":13,"line":0,"coor":0}} [g2]

Works:

N0G0Z0.25000F240.00000
N0G0X0.37500Y0.43750
N0G1Z0.00000F240.00000
N0G2X0.37500Y0.43750I0.00000J-0.06250Z-0.27000F240.00000
N0G2X0.37500Y0.43750I0.00000J-0.06250F240.00000
N0G0Z0.00000

And:

N1G1Z0.25000F120.00000
N2G0X0.37500Y6.44250
N2G2X0.37500Y6.44250I0.00000J-0.06250Z-0.27000F240.00000
N2G0Z0.25000

aldenhart, 2016-01-07:
Gordon, Thanks for reporting this. I may have questions as I diagnose. Should I assume you are using firmware build 086.03 (egde?) Type $fb if in doubt.

gordonab, 2016-01-07:
Ted and I were just discussing this. It appears that I have an old version
of the firmware (83.09):

g2: <-C--1452176113388---- {"r":{"fb":83.09},"f":[1,0,12]} [g2]

aldenhart, 2016-01-07:
The error might still be in there, but I'd like to work form the current edge. Thanks.

aldenhart, 2016-01-07:
Are you getting any messages in the panic? Perhaps something like "Selected plane missing"?

ryansturmer, 2016-01-07:
I can merge the current edge into our branch, but we've stayed behind deliberately to remediate issues caused by changes upstream.  @tedh-shopbot @gordonab - do you think it's a good time to try a new firmware revision?

gordonab, 2016-01-08:
G2.json

{
    "gco": 2,
    "gdi": 0,
    "si": 250,
    "1ma": 0,
    "2ma": 1,
    "3ma": 2,
    "4ma": 3,
    "5ma": 4,
    "6ma": 5,
    "1sa": 2,
    "2sa": 2,
    "3sa": 2,
    "4sa": 2,
    "5sa": 2,
    "6sa": 2,
    "1po": 0,
    "2po": 0,
    "3po": 0,
    "4po": 0,
    "5po": 0,
    "6po": 0,
    "1tr": 0,
    "2tr": 0,
    "3tr": 0,
    "4tr": 0,
    "5tr": 0,
    "6tr": 0,
    "1mi": 10,
    "2mi": 10,
    "3mi": 10,
    "4mi": 10,
    "5mi": 10,
    "6mi": 10,
    "xvm": 360,
    "yvm": 360,
    "zvm": 240,
    "avm": 240,
    "bvm": 360,
    "cvm": 360,
    "xfr": 240,
    "yfr": 240,
    "zfr": 120,
    "afr": 360,
    "bfr": 360,
    "cfr": 360,
    "ja": 0.2,
    "xjm": 200,
    "yjm": 200,
    "zjm": 50,
    "ajm": 20811566,
    "bjm": 20811566,
    "cjm": 20811566,
    "xtm": 0,
    "ytm": 0,
    "ztm": 0,
    "atm": 0,
    "btm": 0,
    "ctm": 0,
    "xtn": 0,
    "ytn": 0,
    "ztn": 0,
    "atn": 0,
    "btn": 0,
    "ctn": 0,
    "g55x": 0,
    "g55y": 0,
    "g55z": 0,
    "g55a": 0,
    "g55b": 0,
    "g55c": 0,
    "ct": 0.001,
    "di1mo": 0,
    "di2mo": 0,
    "di3mo": 0,
    "di4mo": 0,
    "di5mo": 0,
    "di6mo": 0,
    "di7mo": 0,
    "di8mo": 0,
    "di1ac": 0,
    "di2ac": 0,
    "di3ac": 0,
    "di4ac": 0,
    "di5ac": 0,
    "di6ac": 0,
    "di7ac": 0,
    "di8ac": 0,
    "di1fn": 0,
    "di2fn": 0,
    "di3fn": 0,
    "di4fn": 0,
    "di6fn": 0,
    "di7fn": 0,
    "di8fn": 0,
    "gun": 0,
    "xjh": 400,
    "yjh": 400,
    "zjh": 400,
    "ajh": 166,
    "bjh": 166,
    "cjh": 166,
    "xam": 1,
    "yam": 1,
    "zam": 1,
    "aam": 3,
    "bam": 3,
    "cam": 3,
    "1pl": 1,
    "2pl": 1,
    "3pl": 1,
    "4pl": 1,
    "5pl": 1,
    "6pl": 1,
    "1pm": 1,
    "2pm": 1,
    "3pm": 1,
    "4pm": 0,
    "5pm": 0,
    "6pm": 0,
    "xjd": null,
    "yjd": null,
    "zjd": null,
    "ajd": null,
    "bjd": null,
    "cjd": null
}

gordonab, 2016-01-28:
Latest Panic:

g2: --G-1454002967252----> G90
N0G0Z0.25000F240.00000
N0G0X0.37500Y0.43750
N0G1Z0.00000F240.00000
N0G2X0.37500Y0.43750I0.00000J-0.06250Z-0.27000F240.00000
N0G2X0.37500Y0.43750I0.00000J-0.06250F240.00000
N0G0Z0.00000
N1G1Z0.25000F120.00000
N2G0X0.37500Y6.44250
N2G2X0.37500Y6.44250I0.00000J-0.06250Z-0.27000F240.00000
N2G0Z0.25000
N11M30
N12
N13M30
M30

M30 [g2]
g2: <-C--1454002967289---- {"r":{"gc":"G90"},"f":[1,0,4]} [g2]
g2: <-C--1454002967290---- {"r":{"gc":"N0G0Z0.25000F240.00000"},"f":[1,0,23]} [g2]
g2: <-C--1454002967290---- {"r":{"gc":"N0G0X0.37500Y0.43750"},"f":[1,0,21]} [g2]
g2: <-C--1454002967290---- {"r":{"gc":"N0G1Z0.00000F240.00000"},"f":[1,0,23]} [g2]
g2: <-C--1454002967294---- {"r":{"gc":"N0G2X0.37500Y0.43750I0.00000J-0.06250Z-0.27000F240.00000"},"f":[1,0,57]} [g2]
g2: <-C--1454002967310---- {"r":{"gc":"N0G2X0.37500Y0.43750I0.00000J-0.06250F240.00000"},"f":[1,0,48]} [g2]
g2: <-C--1454002967545---- {"sr":{"posy":6.362,"posz":0.250,"vel":100.17,"stat":5}} [g2]
info: Got a machine state change: running [machine]
g2: <-C--1454002967786---- {"sr":{"posy":5.204,"vel":360.00}} [g2]
g2: <-C--1454002968031---- {"sr":{"posy":3.705}} [g2]
g2: <-C--1454002968811---- {"sr":{"posy":2.205}} [g2]
g2: <-C--1454002969046---- {"sr":{"posy":0.773,"vel":253.10}} [g2]
g2: <-C--1454002969269---- {"r":{"gc":"N0G0Z0.00000"},"f":[1,0,13]} [g2]
g2: <-C--1454002969300---- {"sr":{"posy":0.438,"posz":0.245,"vel":12.65}} [g2]
g2: <-C--1454002969549---- {"sr":{"posz":0.015,"vel":26.01}} [g2]
g2: <-C--1454002969607---- {"r":{"gc":"N1G1Z0.25000F120.00000","n":1},"f":[1,0,23]} [g2]
g2: <-C--1454002969696---- {"r":{"gc":"N2G0X0.37500Y6.44250","n":2},"f":[1,0,21]} [g2]
g2: <-C--1454002969758---- {"r":{"gc":"N2G2X0.37500Y6.44250I0.G90","n":2},"f":[1,0,27]} [g2]
g2: <-C--1454002969807---- {"sr":{"posx":0.435,"posy":0.388,"posz":-0.058,"vel":31.98}} [g2]
g2: <-C--1454002969820---- {"r":{"gc":"N0G0Z0.25000F240.00000"},"f":[1,0,23]} [g2]
g2: <-C--1454002969890---- {"r":{"gc":"N0G0X0.37500Y0.43750"},"f":[1,0,21]} [g2]
g2: <-C--1454002969952---- {"r":{"gc":"N0G1Z0.00000F240.00000"},"f":[1,0,23]} [g2]
g2: <-C--1454002970017---- {"r":{"gc":"N0G2X0.37500Y0.43750I0.00000J-0.06250Z-0.27000F240.00000"},"f":[1,0,57]} [g2]
g2: <-C--1454002970057---- {"sr":{"posx":0.370,"posy":0.314,"posz":-0.138,"vel":32.35}} [g2]
g2: <-C--1454002970309---- {"sr":{"posx":0.318,"posy":0.396,"posz":-0.218,"vel":35.01}} [g2]
g2: <-C--1454002970561---- {"sr":{"posx":0.406,"posy":0.426,"posz":-0.270,"vel":25.22}} [g2]
g2: <-C--1454002970809---- {"sr":{"posx":0.428,"posy":0.343,"vel":18.58}} [g2]
g2: <-C--1454002971060---- {"sr":{"posx":0.348,"posy":0.321,"vel":21.27}} [g2]
g2: <-C--1454002971152---- {"r":{"gc":"N0G2X0.37500Y0.43750I0.00000J-0.06250F240.00000"},"f":[1,0,48]} [g2]
g2: <-C--1454002971311---- {"sr":{"posx":0.322,"posy":0.401,"vel":27.79}} [g2]
g2: <-C--1454002971565---- {"sr":{"posx":0.375,"posy":0.438,"posz":-0.248,"vel":35.01}} [g2]
g2: <-C--1454002971812---- {"sr":{"posz":0.179,"vel":83.34,"line":1}} [g2]
g2: <-C--1454002972067---- {"sr":{"posy":0.504,"posz":0.250,"vel":86.69,"line":2}} [g2]
g2: <-C--1454002972312---- {"sr":{"posy":1.622,"vel":360.00}} [g2]
g2: <-C--1454002972560---- {"sr":{"posy":3.124}} [g2]
g2: <-C--1454002972811---- {"sr":{"posy":4.626}} [g2]
g2: <-C--1454002973061---- {"sr":{"posy":6.128}} [g2]
g2: <-C--1454002973135---- {"er":{"fb":83.09,"st":31,"msg":"Move time is NAN - prep isnan"}} [g2]
g2: <-C--1454002973137---- {"r":{"gc":"n0g0z0.00000"},"f":[1,206,13]} [g2]
error: 206,STAT_COMMAND_REJECTED_BY_PANIC,command was not processed because machine is paniced [machine]
g2: <-C--1454002973141---- {"r":{"gc":"n1g1z0.25000f120.00000"},"f":[1,206,23]} [g2]
error: 206,STAT_COMMAND_REJECTED_BY_PANIC,command was not processed because machine is paniced [machine]
g2: <-C--1454002973143---- {"r":{"gc":"n2g0x0.37500y6.44250"},"f":[1,206,21]} [g2]
error: 206,STAT_COMMAND_REJECTED_BY_PANIC,command was not processed because machine is paniced [machine]
g2: <-C--1454002973169---- {"r":{"gc":"n2g2x0.37500y6.44250i0.g90"},"f":[1,206,27]} [g2]
error: 206,STAT_COMMAND_REJECTED_BY_PANIC,command was not processed because machine is paniced [machine]
g2: <-C--1454002973171---- {"r":{"gc":"n0g0z0.25000f240.00000"},"f":[1,206,23]} [g2]
error: 206,STAT_COMMAND_REJECTED_BY_PANIC,command was not processed because machine is paniced [machine]
g2: <-C--1454002973173---- {"r":{"gc":"n0g0x0.37500y0.43750"},"f":[1,206,21]} [g2]
error: 206,STAT_COMMAND_REJECTED_BY_PANIC,command was not processed because machine is paniced [machine]
g2: <-C--1454002973174---- {"r":{"gc":"n0g1z0.00000f240.00000"},"f":[1,206,23]} [g2]
[... 7 lines cut ...]
g2: <-C--1454002973198---- {"er":{"fb":83.09,"st":94,"msg":"Planner assertion failure - buffer empty3 in mp_plan_block_
ist"}} [g2]
g2: <-C--1454002973200---- {"r":{"gc":"n0g2x0.37500y0.43750i0.00000j-0.06250f240.00000"},"f":[1,206,48]} [g2]
error: 206,STAT_COMMAND_REJECTED_BY_PANIC,command was not processed because machine is paniced [machine]
g2: <-C--1454002973202---- {"r":{"gc":"n0g0z0.00000"},"f":[1,206,13]} [g2]
error: 206,STAT_COMMAND_REJECTED_BY_PANIC,command was not processed because machine is paniced [machine]
g2: <-C--1454002973204---- {"r":{"gc":"n1g1z0.25000f120.00000"},"f":[1,206,23]} [g2]
error: 206,STAT_COMMAND_REJECTED_BY_PANIC,command was not processed because machine is paniced [machine]
g2: <-C--1454002973232---- {"r":{"gc":"n2g0x0.37500y6.44250"},"f":[1,206,21]} [g2]
[... cut ...]
error: 206,STAT_COMMAND_REJECTED_BY_PANIC,command was not processed because machine is paniced [machine]
g2: <-C--1454002973245---- {"r":{"gc":"m30"},"f":[1,206,4]} [g2]
error: 206,STAT_COMMAND_REJECTED_BY_PANIC,command was not processed because machine is paniced [machine]
g2: <-C--1454002973246---- {"r":{"gc":"m30"},"f":[1,206,4]} [g2]
error: 206,STAT_COMMAND_REJECTED_BY_PANIC,command was not processed because machine is paniced [machine]
g2: <-C--1454002973320---- {"sr":{"posx":0.000,"posy":0.000,"posz":0.000,"vel":0.00,"stat":13,"line":0,"coor":0}} [g2]

ryansturmer, 2016-02-02:
I am closing this as it is a windows-only issue with node-serialport, and the underlying parse issue it triggers in G2 has been reported to synthetos.

closed 2016-02-02 by ryansturmer