FabMo/FabMo-Engine #303: Problematic Potential Location Bug ... FabMo-SBP-G2 (g55 and "instance") ?

tedh-shopbot, 2016-01-08:
In the process of creating a zeroing system, I have been frequently re-setting an axis zero.

If done a very large number of times ... a strange behavior develops in the axis that is being frequently zeroed -- the ramping seems to get out of sync. That is, there is a stutter at the end of up-ramps and at the start of down-ramps. In the extreme cases, the tool will stall going into and out of ramps, even at modest speeds ... and, in fact, the stutter can be more obvious a slower speeds.

The behavior is specific to the software axis with the exact same moves being smooth on the other axes, When the problem is on the X, if the channel is switched between X and Y, it is the other software axis that will show the behavior (e.g. now on the Y).

By the point that this problem shows up, the g55x value has gone to a large negative number, like -241.685; The x value in instance.json is also large and wierd (I don't understand the relationship of these two at all).

If I save 0 for each of these values and restart the tool -- ramping is normal and the problem has gone away. 

Once one is watching for this issue, you begin to start hearing it after fewer re-zeros ... something seems to be throwing off the start-stop of ramping in the move after some re-zeroing. It is less obvious in moves made with the keypad (composed of small moves) but it can also be detected there.

tedh-shopbot, 2016-01-08:
Since we have been needing to re-zero the X from 30 on all tools, I think we have been starting off with this problem already seeded ...

gordonab, 2016-01-08:
The way the zeroing routines work now in fabmo are very similar to the way
they do in SB3. G55 is set to zero by setting it to the offset to the base
coordinate zero position. So, what is happening is that the base coordinate
system keeps going more negative when you crash into the home stop and
hence makes the G55x number in g2.json a larger negative number each time.
I was going to suggest that you use the VA command, but noticed that the
base coordinate part of it isn't active. I'll play around with this now and
see if the issue goes away.

ryansturmer, 2016-01-08:
A quick guide to the various offset figures found in the files:

1) g55x,g55y,g55z, etc. in g2.json is G2s coordinate offsets for the user coordinate system.  These are the values that are set when you perform a Zero operation (ZX,ZY,ZZ,Z2,Z3,etc)
2) The values in instance.json are just flat out copies of the raw machine position, in table base coordinates, in millimeters, that are stored any time the tool stops moving.  (changes state to paused, stopped, or idle) - this gives the tool memory of its current position, which it restores on powerup using a G28.3.  These coordinates are always in millimeters (while most others change units, keep this in mind)

At startup, the values in the instance.json are sent to the tool in a g28.3 at boot.  The g55 values are similarly sent, so that the tool assumes the last known position when you power cycle.

ryansturmer, 2016-02-08:
Let's get this reproduced so we can fix it for launch - I think we're on the fence as to whether this was the microstep phase issue we saw elsewhere, or if this is a genuine new issue with position loss.

gordonab, 2016-02-08:
I made some fairly major changes last week to the way the sbp position is
handle in the runtime. This needs to be tested further to see if this issue
still remains.

ryansturmer, 2016-02-08:
A fresh look would be good in any case, though I hope that the issue being off by a few thousandths after setting offsets doesn't tie back to the runtime... that would be bad news indeed.

tedh-shopbot, 2016-02-08:
I spent time on this today and was not able to duplicate any issues, either with repeated zeroing in different ways or with plugging strange values in as g55's ... seems to be OK.

closed 2016-02-08 by tedh-shopbot