FabMo/FabMo-Engine #313: No refresh on loading an app ...

tedh-shopbot, 2016-02-03:
When using the app manager to install an app now ... then returning to home, a refresh is required to see the app -- this is new behavior. It does not happen if you use the icon to install the app.

tedh-shopbot, 2016-02-13:
Still not updating on app install within the app manager for me ...

tedh-shopbot, 2016-02-16:
Here are details ...

If, as one of the first things I do, I install an app using the app manager ... it does not show up in the app manager or in the dashboard when I return to the home screen. A refresh is required to get it.

However, if I install one app from the home-screen icon, then when an app is installed in the app manager, it shows up correctly without the refresh. Interestingly, the app loaded from the dashboard-icon does not show up in the app manager until it is refreshed.

After several installs and refreshing, then apps installed from either place seem to go in as expected and do not require a refresh.

ryansturmer, 2016-02-17:
Fixed for good this time, I hope.  You can even install apps in a different browser and they refresh properly.

commit 264db93c (2016-02-08): Fix #313
closed 2016-02-17 by ryansturmer