FabMo/FabMo-Engine #315: wifi wierdness

wlyoung, 2016-02-05:
My Fabmo board always starts in Access point mode since the last image update..sounds like that's expected. If I use the Dashboard network app and select my network from the list, when I enter the password I get a red error message saying "No SSID" and it won't connect, even though I selected the SSID from the list?

If I go to the Updater page though (:9877) and enter that information, it connects to the wifi network OK but eventually falls back to AP mode after a couple of minutes and I have to repeat.

I'm using the image that Ryan installed when I was down there a week ago and the latest Master as of yesterday evening (feb 4). This behaviour started today when I started up.

Bill

ryansturmer, 2016-02-05:
Comments:

1) The AP-mode on boot thing is fixed in the latest.
2) The "No SSID" red message is probably a bug.

Couple of questions:

1) The "No SSID" message - is that on the javascript console?
2) What is the quality of your wifi connection where you have your tool?  Is it kind of spotty?  Or do you have decent signal strength generally, in that location.
3) Had you used the tool before the 4th, (but after the image was administered)? Or is this your first crack at it since you brought it home?

wlyoung, 2016-02-05:
1) I see that message on my phone so assume it's Fabmo-generated.
Screenshot is attached. The network I'm connecting to is "tater". I can
connect OK by filling in the SSID and password in the updater, :9877

2) Wifi is good. There's a repeater in the shop that's right across the
room.

3) I had used it a couple of times to test apps before the Master update
yesterday and the wifi acted as expected. And I connected from the
Dashboard without this error yesterday, when I did the update.

Oh yeah, the dashboard remembers my password, but I get the same error if I
select it or fill it in

Bill

closed 2016-02-10 by ryansturmer