FabMo/FabMo-Engine #320: File Start Dialogue and Confirming Input

tedh-shopbot, 2016-02-09:
Works with green button on Handibot. Future pendant button on all tools ...
Input #5??

gordonab, 2016-02-09:
Input #5 is currently used for all Z prox switches.

tedh-shopbot, 2016-02-09:
What did we hook up the green button to in Handibots?

gordonab, 2016-02-10:
I'm going to have to look at one to figure out what was decided.

ryansturmer, 2016-02-10:
I think as an issue for launch this could use a clearer description - it feels like it might be issues from the title, but could we get a more detailed description of the feature we're looking for here?

tedh-shopbot, 2016-02-10:
As a safety feature, whenever a file is run that starts the router/spindle/drill/etc ... The execution of the file pauses just before the turning on the router/spindle and asks the user to confirm that they want the activation done and the file execution to continue.

This is complicated for two reasons:
1) there is a difference in what happens at the "Start Router?" prompt between alpha tools (where the user hits a separate start button that activates the safety relay and allows the router to come on; followed by a confirming response to the prompt); and a standard tool (where the confirming response just allows the software to continue on and activate the second output switch that turns the tool on)
2) with FabMo, because of the additional safety concern created by the possibility of remote activation, for the software to continue on to activating the router, an input from a switch present on the tool will be needed and required. The case of a standard tool is illustrated by Handibot ... which has an input attached to a green button that is this permissive switch. *New standard tools and existing tools upgrading to FabMo will need to have this safety button added. For an alpha tool, the existing input for starting the tool should be usable for this purpose.

This scenario implies that there is a built in dialogue in the runtime that recognizes when a call to turn on a router or spindle is present in the file ... and serves up the dialogue right after the file starts ... 

Other issues are raised for safety for just doing a move ... and for being able to turn on router for manual cutting with keypad or single commands.

This is all really a new can-of-worms because of the new safety issues raised by FabMo's remote capabilities ... :)

ryansturmer, 2016-02-10:
Thank you, this is perfect.

My proposal would be this:

1. Before any runtime is connected to the machine model (that is, before any motion can happen on the tool) the machine model checks an internal flag to see if motion is authorized.  If motion is authorized, the runtime connection is made, and the tool proceeds as normal.  If not, the machine model enters a "needs authorization" state where it remains until the authorization is given.  The engine can continue to accept commands, it just can't do anything that requires communication with the driver. (G2)
2. The machine model will furnish an `authorize()` function, and it can be called as the result of any number of actions.  For the feature we're discussing, the authorization will be triggered by the green button.  Whenever the green button is pressed, the authorization flag is set.
3. Authorization expires based on a timer.  When the authorization flag is set, a timer is set for a time that is configurable in the engine settings.  (We should discuss whether or not this time can be set to 0 (infinity) or not) - when the time runs out, say 30 seconds, the authorization expires.
4. The authorization timer described in (3) is renewed by any of the following actions:
  1. Reciept of a runtime command over the websocket.  (For now, this mostly means manual tool drive.)
  2. Pressing the green button again
5. Authorization is immediately expired by:
  1. Red button
  2. Tool panic, error, or limit violation

How does this sound?

gordonab, 2016-02-10:
Andrey just clarified that the green button is connected to input #2 on a
handibot.

gordonab, 2016-02-10:
Does the initial green button press remove the need to press it again for a
spindle to start? Or, will authorization need to be given again when the
runtime encounters a Output #1 or #2 for the spindle to start? This isn't
really an issue for a standard, DT or Handibot, but in the case of an alpha
controller the button is tied to an interlocking relay that prevents the
spindle from starting without the button being pressed.

ryansturmer, 2016-02-10:
My thinking is the authorization is for tool use, not for spindle use - you would engage the interlock on an AR at the begninng of the file.

gordonab, 2016-02-10:
So there would still be a message to prompt the user to press the button for the spindle to start?

gordonab, 2016-02-10:
For an AR: As long as Output #4 isn't on when the START button is pressed for authorization we should be able to use it for both functions. This will require adding an additional pair of wires to the pendant and contact block to the switch. We'll need to determine if this is too confusing or not for the operator because they will be pressing the same button twice for different functions.

closed 2016-03-06 by ryansturmer