FabMo/FabMo-Engine #328: Stop Buttons not stopping SBP file ...

tedh-shopbot, 2016-02-20:
Both stop buttons terminate the appearance of the running job in the job manager ... but neither stops the running file (in sb_logo.sbp file).

Also Input #4 -- the red Stop button -- has no effect.

ryansturmer, 2016-02-21:
Ted,

What is the intended behavior of input #4.  For some functions, it simply needs to be configured in the configuration manager - for others, we'll have to discuss implementation.  It will just depend on how you want it to work.

ryansturmer, 2016-02-21:
Also, please post your engine version number (available in the top bar of your configuration)  Does this failure happen if this is the first file you run on tool boot?  (Ie I want to rule out this being the runtime left in a bad state from testing ON INPUT stuff - d1d70ae83b6a3b354af785dc4172e17d05e40b7a makes a fix or two there.)

ryansturmer, 2016-03-01:
The issue for which this ticket is named is resolved, but the physical stop button behavior still needs to be defined.  Will open separate ticket for that.

closed 2016-03-01 by ryansturmer