FabMo/FabMo-Engine #340: Inner frame size on previewer not quite right on linux/chrome

ryansturmer, 2016-02-25:
Scrollbars appear on previewer - seems previewer is slightly larger than the inner frame dimensions.

MaxFrenzy, 2016-05-01:
Noticed this on windows install (chrome) as well.   Also, when in MM mode, I have to zoom back out quite a bit to get the whole previewer (i.e. all 3 axis) in view. This seems to happen if I'm in MM mode in general or if toggling within the previewer.

ryansturmer, 2016-05-02:
Hey @MaxFrenzy can you report your software version? (in the top right corner under "configuration")

MaxFrenzy, 2016-05-03:
@ryansturmer Well at least as far as linux goes (ubuntu), it is software version: ec0f93fdc855-dev. 
There are a variety of other issues that I am experiencing but don't necessarily want to post them as official issues as I am unsure if they are actual bugs, problems with OS/environment, hardware config, etc. I'd like to have a chat with you when you get a chance. -Thanks Matt D.

ryansturmer, 2016-05-04:
Shoot me an email @MaxFrenzy and we can discuss those other issues.  We're looking at this one now.

closed 2017-06-07 by ryansturmer