FabMo/FabMo-Engine #342: Work flow for action on "authorize" ...

tedh-shopbot, 2016-02-26:
- I think it works right for **keypad** at the moment ...

- For the **job manager**, after you authorize with the green button, the file should start, you should not have to go back and click run again (_and in the best of all worlds, you would have a 2 second warning delay with the LED's flashing from turning #4 on, @2sec starting the file; when/if #1 (spindle) is then turned on that should create another 2sec delay to allow the tool to spin-up before motion resumed, LED's will have shifted to slow flashing with both lines on -- in OpenSB the delays are variable and set with SW; we could put them in the machine configuration with the authorize stuff_). I haven't checked how it's set, but what should happen to authorization after a file ends? (after file is stopped??)

- For **Sb4**, if you type in a move and send it, it should run after authorization (rather than requiring retyping as it does now). And, I'm thinking authorization should persist as it does with the keypad (maybe this should be the case for any single line commands or files???)

ryansturmer, 2016-02-29:
Let's open a separate ticket for the LED behaviors, I think that's another discussion.

My worry about making moves immediately after authorization is the following situation:

1. You are not in front of your tool, and you initiate a run.  (This is the situation that authorization is meant to prevent)
2. You are warned that authorization is required, and dismiss the dialog.  "Oh yeah, I gotta hit the green button"
3. You get a phone call, and forget that you have file queued up
4. You approach the tool after your phone call, forgetting that you've got a file ready to go, and you hit the green button (in preparation, perhaps for a manual move)
5. To your dismay, the tool starts running?!

It could be that this isn't a circumstance we're concerned with, but it seems to me one that could come up frequently.  You get an authorization request for an action - you say "ah dang, right" and you go to do something else, forgetting that that action is queued.

The central issue I guess is that the action that is to follow authorization (or not) is not communicated to the user in any way.  A manual move demands authorization but doesn't queue - a job start does.  A "single line" command (Such as is sent by SB4) is sort of a gray area - in an environment where you can connect to the tool from multiple locations and across multiple sessions, the "flow" of the users expereince with the tool can be easily interrupted, and it's really easy to lose track.  I can connect to a tool in the shop, queue up a job, hit run, get bounced for an authorization request, get distracted, do something else, and then come back and hit the buton to find the tool takes off for the corner of the table because that move was banked up.

Maybe a more obvious notion of the tool being "armed" for motion would help?  Or maybe this is a non problem and we should just implement - it seems like a place where folks could get hung up, though.

tedh-shopbot, 2016-02-29:
Good concerns ...
But primary problem is having to select your file twice after your authorize ... or enter a Macro or a command twice, just to get it to execute ...

SO ... if:
    1. requesting authorization initiates the warning state at the tool ... fast blinking red for Handibot ... thus provides warning (device should indicate that you are in this armed state ... I think it does because the request is up)
     2. requesting authorization has an expiration time ... say 10 sec. If no button is pushed, then authorization request is canceled ... thus no "live" tools sitting around.

ryansturmer, 2016-02-29:
So if I initiate an action (Starting a file say) That requires authorization - I enter the `auth` state - this is one of the top level states of the system (alongside `idle` `manual` `running`) - in the auth state, the dialog is displayed, which is really just a warning... you can dismiss the dialog at any time and continue working with the dashboard if you like.  Is the expectation now that that dialog would overtake the display until authorization is granted? By that I mean, the only way to dismiss the dialog is to either click the acknowledge button in the dashboard, which would result in the cancellation of the action, or to grant authorization, which would begin to move the tool?

tedh-shopbot, 2016-04-22:
done or moved

closed 2016-04-22 by tedh-shopbot