FabMo/FabMo-Engine #345: Add support for string type variables

gordonab, 2016-02-29:


gordonab, 2016-02-29:
This code from Macro_90 causes an error:

'MY_VARIABLES.SBC -- File Containing Common User Defined Variables 
'                    Call at the beginning of any file needing to use them

'Tracks whether Shopbot Setup has run
&my_hasrunsetup=0

'Personal info that might help tech support at some point
&my_name=""
&my_email=""
&my_phone=""
&my_purchasedate=""
&my_sbtype=""

'These values are used in the ZZero routine
&my_hasZzero=1
&my_ZzeroThickness=0.121
&my_ZzeroInput=1
&my_ZzeroSafeZ=1.0

ryansturmer, 2016-03-01:
How are string variables meant to be used, in addition to:

1. The argument of PAUSE/MSGBOX command
2. A file argument in FP command

?

gordonab, 2016-03-01:
Can also be used in an if test and,

- write text to a file
- as read by an INPUT

I don't think you're supporting either of these yet.

closed 2016-04-22 by ryansturmer