FabMo/FabMo-Engine #348: Funny JX problem ...

tedh-shopbot, 2016-03-05:
A "JX, 4" causes the tool to move to the right location, but the "FabMo Waiting" message comes up. It eventually goes away and the X location (which looks to be ~4) shows up as 0 in the DRO ... but tool keeps working normally except for JX.
  - J2, 4 (which is functionally the same command) works fine
  - JY, 4 and JZ, 4 work fine
  - And MX, 4 at the same speed also works fine

Some sort of parsing issue? Or, something I am missing??

ryansturmer, 2016-03-06:
Ted,

Please do the following if you can:

1. Repeat the test from a fresh boot or restart of the engine and see if it does the same thing
2. Report your engine version
3. Post any files you have that fail (or is this all interactive commands in SB4?)

tedh-shopbot, 2016-03-06:
fixed by discovering where to find "release candidate" ...

closed 2016-03-06 by tedh-shopbot