FabMo/FabMo-Engine #352: Reviewing STOP behavior and deviations I am seeing from it right now ...

tedh-shopbot, 2016-03-05:
So, per what I think we have all discussed (but who knows with my memory):

1. There will be one type of Dashboard-STOP button and behavior. It will basically work like the current "Pause" button (though marked STOP). That is, the tool will stop with a brisk but controlled ramp (not what is meant by a hard or fast stop). And, then, the user will have the option of RESUME or QUIT. RESUME will require a re-authorization.
2. The physical STOP button (e.g. #4) on a tool like a Handibot will work exactly the same way.
3. This is different than the EMERGENCY-Stop button on full-size tools that will cut power.
4. Users will have the option of using an input to make physical STOP button (in addition to the EMERGENCY-Stop that will be included with their tool) ... in the same way we are using #4.

There has always been some confusion in what these things are called and the use of "stop". What is called a STOP button here (in the Dashboard or on the Handibot) might better be called a "halt" or "pause" button; what is called an EMERGENCY-Stop button might better be called a "kill", "panic", or "stop & terminate all" button. I'm not sure the best way to reduce the confusion. More concerned at the moment about the functionality.

*The following commentary on stop/pause buttons is qualified by the one problem I am having ... (see "Stop Crashing" issue coming next ...)*

For me, **the current "pause" has the expected effect** for what I would want from this button (except, of course, that someday we want it to pull up from the material after it stops). It should have only one name to reduce confusion. I am happy with STOP, but that can be confused  with E-STOP so if there is something else we want to use, go for it. (I haven't liked "pause" because it suggests transience and lack of options; and thus my preference for simply "stop" for this one.)

Similarly, **the physical STOP? button on the Handibot behaves how I would expect at the moment,** and works just like the the dashboard "pause", **IF** it is configured: *Action = Stop; Function = None*.  All other configuration options are producing an assortment of errors for me. But, I do think this configuration makes sense in terms of Alden's talk about input functionality.

ryansturmer, 2016-03-06:
This all makes sense, but some of it is substantial work to implement (in the OpenSBP runtime - easy in g-code)

Below, I will continue to use "pause" to describe what we want to relabel, perhaps simply to STOP, only because that's how it's called in the code currently:  (Orange button)

When the system enters the "paused" state - what do we want the UI to do?  Options:

1. What it does now:  Unceremoniously replace the "pause" button with "resume" and "quit" buttons
2. Modal dialog: Display a modal dialog with the same options

The advantage of 1. above is that you can continue to work with your dashboard (use apps, etc) unencumbered by the tool stoppage.  The disadvantage is that it doesn't jump out at you that your tool has stopped (although you presumably know this if you're looking at your tool.) 

The advantage of 2. above is that it really gets in your face about having to address the issue of your tool having paused, but locks up the dashboard for any other options (including, down the road, the ability to drive the tool manually perhaps, during a pause)

tedh-shopbot, 2016-03-06:
I believe that #1 is the best ... there are people who do a lot of this Stop>Resume activity (just to re-assure themselves that things are going ok and that they are in control). So I am happy with the functionality of this from the dashboard, except for how it is labeled, which as noted previously ... I am not sure what word is best, but pause is my least favorite.

And, #4input seemed to be doing the same thing (as I configured it) which is, I believe, what it should do.

ryansturmer, 2016-03-07:
The work in the changes is in the required reauthorization and workflow for runtimes.  I think we're all on the same page here.

tedh-shopbot, 2016-04-22:
implemented I think ...

closed 2016-04-22 by tedh-shopbot