FabMo/FabMo-Engine #353: "Stop" Crashing ...

tedh-shopbot, 2016-03-05:
Generally, the "pause" button (my notion of "stop") works well. Using the physical Stop button on the Handibot (#4) has the correct identical effect (when configured action=stop/function=none).

However, I found when using these stops repetitively that they occasionally just locked the tool up with the display showing persisting with the usual "Pause/Stop" option as if it was still running. I was getting this by just stopping at various points in the sblogo.sbp file. Eventually, I noted that I did not get the problem if the tool was obviously making a move when I stopped it. It seemed that the crash happened if I was between stacks when the stop was requested (this was true both for the #4 button and the Pause button in the Dashboard).

I made a test file that was just multiple MX, 5 --> MX, 0 moves. 
  - I first put a 2sec pause at each end of the moves to break the stack. Running this one, I was not able to get the crash with dozens of pause/resumes at various places.
  - Then I just put a simple equation between each move to break the stack. Now, if I hit either type of stop just as the move was terminating ... I got the crash (one that seems to require a reboot).

That is, it seemed like that the problem was happening when a stop was triggered at the time that nothing was going on. Stop seemed to be picked up in a move and picked up in a pause. But, it was pretty easy to get in between ... could this all be something new because jogs (in the logo file) are breaking the stack in ways they did not previously do?

ryansturmer, 2016-03-06:
Ted, please report engine version - I assume release candidate?

PAUSE does not break the stack, and depending on the equation you used, it may not either.  (Expression evaluation *should* only break the stack if it demands the read or write of a stack breaking variable)

Its possible that jogs are the cause. They are breaking the stack in master, when they shouldn't - I took that change out in the rc.  If you swich you should see it.

In either case, stack breaks shouldn't cause stop/pause issues, so if that's the case, we can get it resolved.  Looking into it...

ryansturmer, 2016-03-07:
Fixed in latest 92ed5c

closed 2016-03-07 by ryansturmer