FabMo/FabMo-Engine #354: Dont' know if this is intended or not ... editor behavior

tedh-shopbot, 2016-03-07:
If you open a Macro to look at it ... and just exit back to home because you have not done anything. Then, if you open the editor app you get the last macro and the macro menu.

ryansturmer, 2016-03-07:
This is the case for all apps - when you open the main menu, you're not actually closing the app you were using, you're just hiding it.  We should fix this, probably, since it produces behaviors like this that are unexpected.

tedh-shopbot, 2016-04-22:
What this effectively does though, is make you feel like once you have opened the macro editor, you can't open the regular editor. It does clear/restore on a refresh ... but that may not be obvious to some. Probably not a big deal, but it might be nice to always refresh the editor after a macro exit ...

ryansturmer, 2016-04-22:
No this is definitely a big deal.  I think I will probably have the dash unload an app no matter what, whether you go back to something that's *integral* to the dash, or something that's just another app.

tedh-shopbot, 2016-04-22:
I do sort of like that the regular editor does not forget though ... let's you do a little something and then come back and do again ... w/o working through job manager ...

commit 4c103697 (2016-04-29): Close #354
closed 2016-04-29 by ryansturmer