FabMo/FabMo-Engine #362: Pause Problem in Macros ...

tedh-shopbot, 2016-03-17:
If you have just used the keypad and given an authorization (say several min)
-then if you run macro 2 that has a pause for continuation input;
1. the macro starts without request for authorization (that is probably ok) 
2. but then the Pause for input, when encountered, also runs right by without stopping at the pause ...

tedh-shopbot, 2016-03-17:
Also ... in continuing to test this ... I've snapped the clip on the bit as it plunges ... and while it stops on contact, the file no longer continues. Don't know if this is a related problem, but the touch-off, with repeat touch-off, had been working pretty well earlier.

ryansturmer, 2016-03-18:
1. What version of code?
2. How are you launching macro 2.
3. When you say pause for input - do you mean pause for input switch or pause for message?  Which line in C2 are you referring to?

tedh-shopbot, 2016-03-18:
1. It was with the development version that I updated to just before testing at about 4 this afternoon ... I left that FabMo at Handiland ... but am installing another to get number and re-check now ...
2. Did the same thing when launched from and app and when launched from the Macro window. I had used keypad just before and given authorization so it did not ask when starting Macro 2, in either case.
3. Pause as in the PAUSE statement at line

ryansturmer, 2016-03-18:
Try the latest dev now, if you're testing, I have made some stability improvements related to authorization and machine state.

tedh-shopbot, 2016-03-18:
Just updated ... v0.0.7-69-g15854b5

tedh-shopbot, 2016-03-18:
OK the PAUSE seems to be working right now. Stopped from both macro-window start and app start. I don't have a unit set up to check the #1 touch-off input ... so that will have to wait until tomorrow.

For future consideration:
- don't know if this is what you expect, but in the condition where an authorization was recently given for a manual move and a long timer was in play, when run from Macro window, re-authorization was requested, but when run from app, same macro ran without re-authorization
- at some point we may want to consider as behavior for pause:
    = continuation produced by green button only
    = quit produced by either red button or click on quit
  ... this would simplify the responding.

ryansturmer, 2016-03-18:
1) Yes, this makes sense, in your app, your'e not calling the fabmo runMacro function, you're running an immediate command.  I can make runMacro use the same auth strategy as immediate commands - your call.
2) Yes, I agree this would be good.  Probably not for launch, but we can do that for sure.

tedh-shopbot, 2016-04-22:
seems mostly dealt with ...

closed 2016-04-22 by tedh-shopbot