FabMo/FabMo-Engine #365: Lockup Errors from standard V-Carve post with OpenSBP

tedh-shopbot, 2016-03-19:
[test_carve_x2.txt](https://github.com/FabMo/FabMo-Engine/files/180784/test_carve_x2.txt)

The attached is a small SBP file that is in the standard output post from V-Carve.
There are two lines, marked with '***** that each produce a lock-up failure if normally exposed
File runs as expected with the lines commented out ...
Recovery requires restart of FabMo.
V1.0.0-6g037018c RC am 1/19
Logs for clean and error runs via email

gordonab, 2016-03-19:
I'll check on the SA command, will have to experiment with the IF line.

tedh-shopbot, 2016-03-19:
Yeah ... I don't know if the IF is the problem or the call to the system variable ... but since everyone's files are going to have these lines in them, it is a bit of a problem.

ryansturmer, 2016-03-19:
Latest RC fixes a bug, but not the IF issue - this is related to recent changes we've made in what breaks the stack vs what doesn't.

closed 2016-04-22 by tedh-shopbot