FabMo/FabMo-Engine #369: Problem with Jimmy's new SBP jobs ...

tedh-shopbot, 2016-03-21:
Hey Gordon, could you take a look at this ...

- Jimmy created a nice test file from aspire
- when posted it g-code it runs fine
- when is SBP it reveals several issues ...
- there is an error at start related to the ShopBot calls to macros, etc ... (MORE BELOW)
- But, if you cut off the top and bottom, the file will start, and then locks up ... too much SBP Code? 20K lines??
- If I make a very short version from that ... it runs ...
- My own test file with similar posts and that calls the same macros seems to run ok
- However, I did just notice that macro90 does not run right from the macro manager ... and it should
- all the current macros are archived in the current RC
- *I'm going to email these job files for expediency ...

ryansturmer, 2016-03-21:
1. Large SBP files are going to hang up the system at present, what was the behavior you observed when you ran the file?
2. When you say "does not run right from the macro manager" what is the behavior vs. what is expected.

Note: at present, large SBP files are going to be slow to start/run.

tedh-shopbot, 2016-03-21:
When you directly run macro90 from the macro manager, it generates an error. I have not been able to relate anything to the particulars of the message. Thought if someone else took a look at it, it might make more sense and I could get back to docs ...

ryansturmer, 2016-03-21:
My C90 fails as "unexpected quote on line 26" which corresponds to an assignment of a string to a user variable, which is currently illegal.  No string type user vars yet.  There's an open issue for this, but for now, a file with string type user vars won't even parse.

tedh-shopbot, 2016-03-21:
Hmmm ... what is large for SBP ?

ryansturmer, 2016-03-21:
More than a few thousand lines, probably.

tedh-shopbot, 2016-03-21:
Yep ... that's the Macro90 error. I've been calling that macro from a file for the last week or two. It does not seem to notice from within the file. Only in trying to debug Jimmy's crash did I run it directly. Note that I have been using it to set up some Handibot specific stuff. For present purposes we can just delete all those strings in the macro.

tedh-shopbot, 2016-03-21:
So that means that all is explained ... no problem Gordon.

ryansturmer, 2016-03-21:
To be sure, I would run the file again, (fix the issue with C90) and give it time to load and run.  It may take several minutes to get going but should work fine once it's started.  If you wait and wait and it doesn't start - then we still have a problem.   If you're looking to just get the file working though, a g-code post will do the trick for big files.

tedh-shopbot, 2016-04-22:
now subsumed under other issues and testing ...

closed 2016-04-22 by tedh-shopbot