FabMo/FabMo-Engine #373: 2 Pauses with messages in succession not executing the second

gordonab, 2016-03-22:
Scott Worden reported that when he puts 2 pauses in a row the second one is ignored. He is trying to tell the operator that the spindle will start and then tell them to manually change the VFD to the speed in the file. He has sent a sample file (attached, change the file type to SBP to use in FabMo).
[Pause test.txt](https://github.com/FabMo/FabMo-Engine/files/184473/Pause.test.txt)

ryansturmer, 2016-04-22:
This appears to be fixed, engine-wise but provokes some sort of slowdown in the dashboard that we should examine.

wholetthedogsout, 2016-04-25:
I have been experimenting with this, and I seem to be getting the expected events, possible we were running a version  without the unbinding? @ryansturmer

ryansturmer, 2016-04-26:
This was latest master on my laptop - so I think unlikely, but I'll try and run it again and see

closed 2016-04-26 by wholetthedogsout