FabMo/FabMo-Engine #378: When Paused-Dialog in Macro ... quit/resume inappropriately displayed at bottom

tedh-shopbot, 2016-03-24:


ryansturmer, 2016-03-24:
What is the appropriate behavior here?  My understanding of a pause was that it should provide both of those options, eg:  "Make sure you have a bit in the tool" and you get the choice of stopping the file if you need to move the tool around, or you can resume if you've installed the bit.  Are there sometimes when we only want to allow a quit or a resume?

tedh-shopbot, 2016-03-24:
The behavior is right. But in the dialogue at the top where the PAUSE question/dialog is served up, the user is asked to continue or exit, and given buttons there ... but then down below, there are these other buttons that say resume or quit that are the buttons they have seen in a Stop situation. It seems distractingly redundant and confusing ... even though the options probably have the right effect.

ryansturmer, 2016-03-24:
Ah!  I vote that we leave those in for now, until @wholetthedogsout get's the new modals on the water.  There are times when the modal dialog should be showing but *isn't* and in those instances, the buttons at the bottom are the only recourse.  I agree though, it's redundant and confusing, I just hadn't noticed it!

tedh-shopbot, 2016-03-24:
Maybe we could shift to using the same words above and below ... e.g. resume = resume (rather than continue) ...

wholetthedogsout, 2016-04-18:
We can shift this entirely over to the modal now? @ryansturmer @tedh-shopbot

ryansturmer, 2016-04-18:
I think that would be fine, but at least @tedh-shopbot should weigh in

tedh-shopbot, 2016-04-18:
I think so ...

closed 2016-04-22 by tedh-shopbot