FabMo/FabMo-Engine #385: Config/Offset not applied to .nc files

SW0rden, 2016-03-29:
I forgot that I had offsetting enabled and while trying to figure out why my .sbp file was cutting -2.0" in X, I tried a .nc file and found that g-code files are unaffected by offsetting (preview also).

Offsetting also affects the z-zeroing routine.  This was catching me up a bit too, because of my SB3 conditioning where offsetting is canceled upon completion of the file being run.  This also affects commands in the SB4 app.

ryansturmer, 2016-03-29:
Scott, what do you mean by "ofsetting enabled?"

gordonab, 2016-03-29:
If Offset is Enabled in the configuration it won't effect a G-code file when run. Right now the Offset, Scale, Rotate... transformations only apply to OpenSBP code.

SW0rden, 2016-03-29:
Enabling offset in config will run a file offset from the current 0,0,0 location.  I had done this last week and forgot to 'Disable' it before running a file yesterday.  After Z zeroing with the macro, I couldn't figure out why my z raised to 1.01 each time instead of the commanded final safe z position of 1.0.  I had a x offset of -2 and a z offset of 0.01 that was being applied and didn't realize it until previewing the file as .sbp and .nc and .sbp files will show in preview and run with the offset, but .nc files will not show the offset in preview and they also run without the offset.

[image]

SW0rden, 2016-03-29:
Ok, thanks Gordon, that explains it.

Just a note, the value box labels for rotation, scale, and offset stay as mm when the default unit is set to inch.

SW0rden, 2016-03-29:
Sorry if I shouldn't have hit "closed".  Not sure of the 'etiquette' here.

gordonab, 2016-03-29:
I will check on the unit types and make sure they're set up correctly in the html file.

closed 2016-03-29 by gordonab

ryansturmer, 2016-03-29:
Ofsetting is a feature of SBP, and will not work with g-code files.  This is by design.  We will make this clear in the documentation.