FabMo/FabMo-Engine #393: Network Revert

ryansturmer, 2016-04-04:
Revert and Return. The elephant is, of course, the user not knowing the FabMo IP on the network that they were trying to join (for whatever reason) -- and not having a wide enough range of tools available to figure it out. SUGGESTION: Adding to the Network Manager a: Return to Last Network option that also displays the IP address (noting that the address will typically persist for several days, and that they can use it to login). Clicking this new feature would return FabMo to that network (without requiring the pwd). Having a simple and confirmed method for getting to AP mode, and a quick way to retry getting back on the network, now with the IP, will eliminate much pain. The new area in the Network Manager might even contain additional info about that last network (or failed attempt) that could help diagnostically in situations where getting on a network is not working.

ryansturmer, 2016-04-04:
Couple of issues:

1. If authorization is disabled, there effectively *is* no green button.  Should I track authorization enable/disable separately from the input used for auth?  I'm thinking probably so?

2. There is no special distinction for the red button - you can label any button as a stop button, so we need to track this separately in the machine model.  What happens if you designate more than one button as the stop button?  Which one is used for this feature?  What happens if you designate a stop button, but it gets set such that its action is not to stop the tool?  Will this be enforced by the machine model, or that input's configuration suppressed?

3. An alternative could be that 5-10 secs on the auth button alone is enough to kick it to AP - is that an acceptable solution?

gordonab, 2016-04-04:
I like the idea of requiring 2 button be pressed in order to make the switch. This should help inadvertently switching from station mode to AP in cases like someone laying a piece of wood or something on top of their Handibot when they take a break.

Would this be a good time to shift the green and red buttons to inputs that aren't normally used? This may take care of some of the questions about changing the type of designation and confusion when we start putting Fabmo on the other machines.

ryansturmer, 2016-04-04:
I think the issue of input designation (whether frequently used or not) is going to keep coming up - right now, you can set an input as the auth button, but also assign it as an action button for G2, which I think is not ideal - one way to proceed is to just wrap up all the G2 stuff, and don't allow direct access to it - basically wrapping every input with a "fabmo input type" that includes higher level functions like "Authorization Button" and "Stop Button" - but also includes low level things like generic input, etc.  This complicates things somewhat from the bookkeeping side in the engine, but makes it simpler for the users, which I think is what we want here.

gordonab, 2016-04-04:
How does this flexibility in which inputs are used for Authorization
prevent the user from designating none as this function and circumventing
this system? Will they always have to have at least one designated for the
Authorization button and one for the Stop Button? The idea of making any of
the inputs capable of the this functionality makes sense and could possibly
improve the user experience in the long run by simplifying the decision on
which input to use.

ryansturmer, 2016-04-04:
I think people always have the option to not specify an authorization button, not specify a stop button, or both.  You're always going to have the development user who isn't interested in either physical button.  Those people forfeit the ability to trigger an AP collapse with the buttons.  Presumably, if you know enough to want to disable the safety features of the tool - you know how to get your tool on and off the network?

tedh-shopbot, 2016-04-04:
Just catching up on this discussion ... I'm good with the gist.

SW0rden, 2016-04-04:
Just my 2¢ worth....I don't like seeing user inputs/outputs used for system input/outputs as this takes functionality away from us.  Meaning, I'll have less to use for my needs, if desired, as time goes on.  I'd rather see these internal, but still manageable.

Instead of doubling up on button functions, what about a 'reset/reboot to AP' button that is 'under the hood' so to speak?

ryansturmer, 2016-04-05:
There is already such a button, @SW0rden , but the operation is (at least at this time) common enough to want to do something that's more convenient than opening the hood and accessing a tiny button.  I hope this changes as time goes on, but as it is, it's valuable to be able to double up in this way on a platform such as (eg) the Handibot which only has two buttons that are routed to the chassis.

SW0rden, 2016-04-05:
Oh yeah, I know that button well.  Pushed one right off the board.  No, I meant a small 10mm or so button routed to an internal location just under a cover, or better maybe, a flush mount (slightly below flush) that couldn't be accidentally depressed instead of one of the larger buttons.  Labeled 'Reset to AP' or something.

gordonab, 2016-04-05:
This doesn't help now, but the spec on the new Edison board is slated to
have a bigger button that's accessible when you take cover off.

tedh-shopbot, 2016-04-05:
Just to clarify my thinking on this a bit ... I don't see the functionality I am suggesting here as some sort of special, protected thing you do to reset/restore your operating system when you are in distress. Rather, I see it as a normal work-flow "feature" for FabMo that lets you conveniently move between networks and offers a basic tool for discovering the network info you need to get onto a network.

In fact, it presumes that the engine is working normally and just helping you out a little, either getting back to direct access, or getting back and letting you know what you may need to know about a network to get on it. Thus, I believe it should use the normal buttons and be viewed as relatively normal behavior and convenient functionality, not require finding a strange little hole to poke at.

ryansturmer, 2016-04-05:
Agree @tedh-shopbot - Although I am hoping that while this should be seen as a "normal" thing to do - my hope is that over time, the performance of the network management in the dash, combined with the tool minder on its various platforms, makes this something you don't have to do often.  It's certainly a clunky and slow way of going about things, and we'd be better off if we limited the amount that a user had to be hopping back and forth to AP mode.

ryansturmer, 2016-04-05:
And disagree with @SW0rden - Adding these enhancements to the buttons doesn't somehow take away their functionality as generic inputs, it simply furnishes more options for how inputs can be used, either by the end user, the implementor or the tinkerer type developer.

tedh-shopbot, 2016-04-05:
@ryansturmer Agreed on both. For me, it's largely giving the user the confidence of knowing there is always a route of access. Say, you are in your shop with a new phone ... the Handibot jumps onto the network ... your phone has not got a toolminder or anything on it and you never bothered to look at the IP when you used the tool minder on another device ... JUST "revert and return" and you're in business. Use the App on the tool to get you to where you can get the toolminder for that device.

ryansturmer, 2016-04-15:
There is a "network history" in the network manager now that shows the latest networks connected, and you can hold the green button for 5 seconds to trigger an AP collapse, so between the two of these features, there's always a recourse for users who can't find tools on their network.  The workflow goes like:

- Connect in AP
- Join the requested network
- Search for tool with tool minder on network
- If you can't find the tool that way, hit the green button for 5 seconds
- Connect in AP and visit the network manager
- The network you just joined will be in the history - you can click it to re-join, confident now that you know the IP address

closed 2016-04-15 by ryansturmer