FabMo/FabMo-Engine #396: Arrow keys sometimes "sticky" in mobile

wlyoung, 2016-04-07:
On occasion if you press a key in the keypad on a mobile device, the arrow "sticks" and the movement continues after you release the arrow. The arrow stays green and you have to click again to stop the motion.

Unfortunately I can't duplicate on demand but it happened on an ipad this morning in Z. Newest linux and 1.0.3 engine, and running chrome on the IPad

ryansturmer, 2016-04-07:
ipad only?  Or all mobile?  Seems like you mostly observe it on ipad?

SW0rden, 2016-04-07:
I have seen this on an Android tablet with FireFox, but not Chrome with the latest image and engine as well.  It is 100% reproducible for me.

wlyoung, 2016-04-07:
I've also had it happen in the past on android phones with Chrome, both a Galaxy S5 with 1-version-back android and a Motorola Moto g with the latest Android. Will test with Firefox in a little bit.

ryansturmer, 2016-04-22:
I observed this at least once when the authorization dialog popped up - were you using authorization bill on any of these tools?

ryansturmer, 2016-05-23:
This is fixed afaik.  We can re-open if not.

closed 2016-05-23 by ryansturmer

SW0rden, 2016-05-23:
Still see it on an Android tablet in FF only.  Could just be the older Nexus 10 tablet, but it is running version 5.1.1.

ryansturmer, 2016-05-24:
What version are you running SW0rden?

SW0rden, 2016-05-24:
Software Version
v1.1.1-1-g2b3fe2c