FabMo/FabMo-Engine #399: In 1.0.3 neither MS or VS works in a file, in the editor, or in Sb4 ...

tedh-shopbot, 2016-04-09:


gordonab, 2016-04-11:
Were the MS & VS non-functional or non-persistent? I couldn't tell with my simulator, so I have to try this here this morning on a real machine. Did you try the SV command to make these speed changes persist?

tedh-shopbot, 2016-04-11:
I guess my first issue was making them work from the command line in  Sb4 ... no effect of either. Then, I tried in a file and it did not seem to change the speed.

OK, I'm remembering ... I was going to change the Sb4 to issue an SV before sending a speed change???

gordonab, 2016-04-11:
That is correct. Let me know if that still doesn't work.

ryansturmer, 2016-04-15:
MS and VS both work to change the speed in a file.  VS does not appear to persist. (latest RC)

ryansturmer, 2016-04-15:
Let's keep the use of version numbers out of the titles of issues, since technically, version 1.0.3 will *always* have this issue :)  - It is however fixed in the latest rc.  You need to issue an SV though, to cause values to persist.

closed 2016-04-15 by ryansturmer