FabMo/FabMo-Engine #402: Odd jerk values for A, B and C axes being saved to G2.json

gordonab, 2016-04-12:


ryansturmer, 2016-04-12:
To reproduce: Set a value with MS for the x,y, or z axes, do a VS, and then check G2.json

SW0rden, 2016-04-12:
@ryansturmer, would this be related to the issue I had with abrupt jerk stops/lost steps, and after deleting the two files you ask me to delete (G2.json and instance.json), it was cleared?  This crept back in on me again today and after deleting them an rebooting again, it was gone.  I've run files with MS, but haven't issued a VS at any time.  I did however attempt to set/try several different min/max velocity settings as well as jerk settings before it started.

ryansturmer, 2016-04-12:
Scott, the next time you're prepared to delete those files, it would be instructive if you could save them, so that we can understand if there are actually settings in there that are going haywire on you.

SW0rden, 2016-04-12:
I didn't save them out this time, but will if it happens again.  You do have the last set that was problematic (think I have them somewhere too yet).

tedh-shopbot, 2016-06-02:
Just wanted to update this one. Odd Jerks and Speeds in the new unit value branch for A, B, and C. Change when you enter, sometimes when you restart ... and strange numbers ...

SW0rden, 2016-06-03:
One thing I've noticed with the value fields throughout config is that when I click in the field and swipe to select the current value, type the new value, click and swipe in the next field, and by the time I type the value in, the selection has vanished and I've now typed in additional values to the end or beginning of the current value.  This has caught me a few times and I now wait a few seconds before clicking and swiping to select the next field value.  If one wasn't paying attention to what the final value is, you could have a undesired value entered.  For example, a jerk value of 100 intending to be changed to 60 could end up being 10060 if you didn't catch it before continuing on.

ryansturmer, 2018-02-06:
I believe this to be fixed in latest dev.

closed 2018-02-06 by ryansturmer