FabMo/FabMo-Engine #405: Up/Down DRO arrows in FF

SW0rden, 2016-04-15:
Seeing this in FF, but not Chrome on PC only.    

v1.0.3-50-gd0af424  

[image]

SW0rden, 2016-04-15:
Should note, that in Chrome, they show only when moused over.  I assume that is the intended way.

wholetthedogsout, 2016-04-18:
This is an addition by the browser for number inputs, you cannot alter them, but can eliminate them. I find them kinda useful but we can get rid of them if people don't like them.

SW0rden, 2016-04-18:
Is it the case that in FF they're always visible?  In Chrome, they only show when moused over.  Note again, this is only with a PC and not a tablet.

I don't mind them as long as they're hidden until moused over like in Chrome.  I'd find them more useful if you could set the increment to less than an inch at a time though.  Would really like to see some form of Move Nudge implementation.  

I see that once used in FF, the three places after the decimal point are dropped.  I take it that FF is a bit more of a problem child compared to Chrome?  Fonts are sharper in FF, but the overall experience is better in Chrome.

ryansturmer, 2016-04-18:
You can nudge the tool by tapping the buttons scott - the nudge distance is configurable in the machine settings.

FF vs Chrome is a toss up - I find that the system is generally faster in firefox, even though we do more testing in chrome than in FF.  Neither browser is perfect, and if FF seems like a "problem child" it's probably because we're not doing enough testing.

SW0rden, 2016-04-18:
Oh yeah, I have that set and use it often for sneaking up on a position.  I just like the MN command in SB3 which nudges and then resets the axis accordingly/automatically.  The two are different.

I have noticed that FF is faster.  I kind of assumed you were more Chrome centric there though, and that is where I've been using it mainly, but I do take a look at it in FF from time to time for comparison sake.

ryansturmer, 2016-04-22:
What's the actionable item here?  Do we want to remove these?  do we want these to behave the same across browsers?

closed 2016-04-26 by wholetthedogsout