FabMo/FabMo-Engine #418: Need function to get whether internet is accessible or to detect AP mode ...

tedh-shopbot, 2016-05-10:
For system to select between local and cloud documents, need to know whether connected to internet ... but it may be good enough just to know if we are in AP mode (since we clearly are not connected in that case)

ryansturmer, 2016-05-10:
I think it's probably insufficient to use the AP mode flag - after all knowing that the tool can't see the internet doesn't mean the client device can't - I'm sure we can do something here, though.

ryansturmer, 2016-05-10:
Here's a follow-on question:  If you have a local document, why would you ever try to do the cloud document, unless you believed the cloud document to be more up to date, in which case, you probably shouldn't without warning display a document you know to be out of date?

tedh-shopbot, 2016-05-10:
Question 1 ... yes, but generally when you are in AP you will not have internet unless you've got your PC on ethernet or some sort of phone connex I don't have.
Question 2. Fair question. In some cases, it might be better to have a short stand-in document available, just to have one available. In the case of something like the "news" we might make some other sort of comment. In the new day of Apps ... I expect we'll always be more up to date and maybe having a local document is always better. But, I also have help/resource links to websites and forums and it would be nice to gray them out if there is no connection.

tedh-shopbot, 2016-05-10:
I can skip all this ... was just fussing with stubbing in all my old functionality. Next you will be complaining because I want to stream the lines of the file :)

ryansturmer, 2016-05-11:
These are all valid, I'm just working through to get the best solution.

Why not something like this:

```js
var remoteURL = 'http://docs.handibot.com/doc-output/Handibot2_Safety.pdf';
var localURL = 'files/joining_network.pdf';
$.ajax({
	url:remoteURL,
	type:'HEAD',
	error: function() { fabmo.navigate(localURL); },
	success: function() { fabmo.navigate(remoteURL); }
});
```

ryansturmer, 2016-05-11:
This doesn't obviate the need to detect AP mode, which can be done, but could use a convenience function, but that's a separate issue I feel, from this.

jlucidar, 2016-05-11:
it triggers a cross origin issue :) (we already tried this)

ryansturmer, 2016-05-11:
I tried this on my machine without issue, I must be running with some elevated permission or something?

ryansturmer, 2016-05-11:
Attached FMA with the above proposal:  I run this in my dashboard (in the iFrame) without issue - can someone try it out and see if theirs produces an XSS issue?

[example.zip](https://github.com/FabMo/FabMo-Engine/files/258620/example.zip)

ryansturmer, 2016-05-11:
Maybe this worked because i was running on localhost?

ryansturmer, 2016-05-23:
fabmo.isOnline() will work for determining if the tool is online, and you can use the above technique (with an XSS permissive host) for determining if the device is online.

closed 2016-05-23 by ryansturmer