FabMo/FabMo-Engine #420: User Authentication

ryansturmer, 2016-05-12:
Authentication per the auth branch.

ryansturmer, 2016-05-24:
Things left to do here:

1. Merge the root endpoint and authentication endpoints so there's no redirecting, etc.
2. Incorporate user account management into the configuration app
3. Protect the new user creation endpoint and remove the button from the sign-in page that creates a new user

jlucidar, 2016-05-28:
points 2 & 3 are done (see  dc0ece1ee455bbb0386fb66f3e1be16705c20a0c) ! 
How do we want to integrate point n°1 (to replace [that implementation](https://github.com/FabMo/FabMo-Engine/blob/authentication/authentication/authentication.html)) ?  

@wholetthedogsout, @ryansturmer  any idea ??

closed 2016-06-16 by jlucidar