FabMo/FabMo-Engine #434: Shortcut to bring up Keyboard Arrow Key mode ... also ...

tedh-shopbot, 2016-06-03:
To be able, whenever DRO is up, to use a keyboard shortcut key to activate keyboard mode ... for example, the letter K ... The idea here is that if you don't want to use a mouse to move the tool, you probably don't want to have to use a mouse to enable using you keyboard.

The second part of this, is to be able to make the same call within SB4 with [SK] full command; or 'K', the highly used shortcut ...

And, naturally, we also want to short press on keyboard keys to have the same discrete move as the short press on the touchpad screen.

SW0rden, 2016-06-07:
I'd like to add to this, the ability to shift to jog mode, normally ctrl+arrow for keyboard and right click on keypad arrows.

ryansturmer, 2016-06-07:
How do you want to address the issue of the letter K being pressed in apps, when the DRO is displayed? (For instance in filling in a parameter field)

tedh-shopbot, 2016-06-07:
Hmmm ... when an app is up, the UI no longer supports the key; but the app can optionally feed the call to the functionality. So in Sb4, the app would intercept the key when appropriate and make the call.

I am actually OK with it only working that way. I just thought some might want to have it available automatically if in the simple DRO only situation. I guess the challenge is making sure that we have keyboard input and not touchscreen input. I'm assuming we have a way to detect that??

ryansturmer, 2016-06-07:
We dont easily, unfortunately, but I'm sure we can make that work somehow.

-R

ryansturmer, 2016-08-03:
Let's revisit this - @wholetthedogsout when you get the chance, could you bind a key event in the dashboard to starting the keypad controls?

closed 2018-12-06 by tedh-shopbot