FabMo/FabMo-Engine #435: DRO; defined distance moves -- lock?

tedh-shopbot, 2016-06-03:
I have been playing a lot with little moves to make adjustments lately ... When one has a perfect connection the ones achieved by just tapping the direction keys are very convenient; however, in most situations, whether you get a move or not, or 3, seems pretty unreliable. I think we should consider locking fixed moves either on or off so that a push, even if held too long, produces one fixed distance move. This is the only way you can really use if for something like carefully zeroing a cutter ...

And, naturally, I'd like it to work the same way in keypad and keyboard.

Where we store/change the distance and where we turn it on and off ... makes it all more complicated. But this is a function that if not 100% reliable is not very useful.

ryansturmer, 2016-06-21:
There are a couple of ways to tackle this:

1. We can keep it as is and adjust the time that is considered a "tap" (we can lengthen that) so that it's more forgiving what counts as a fixed move.
2. We can add a "lock" feature to the keypad which when turned on will result in the behavior you describe.

Fixed distances are already adjustable, in the machine settings - I vote we leave them in there for the time being.

tedh-shopbot, 2016-06-22:
I think having them adjustable where they are is fine. Solution number one would make keypad feel even less responsive, and I don't think it would solve the problem. My own feeling is that locking fixed distance on is the only robust solution. Yes ... an extra tap, but when you are trying to manually z-zero, which is the primary use, uncontrolled/unresponsive plunges are a drag.

I've always been happy with it working this way in Sb3, and have never had a complaint. Run-away keypad moves, however, get people really upset (and we are not free of them yet).

ryansturmer, 2016-06-22:
Can we use this ticket as an opportunity to discuss the location of the lock button?  I think a mock-up before we implement is appropriate.

closed 2018-12-06 by tedh-shopbot