FabMo/FabMo-Engine #438: Update to RC Failed

SW0rden, 2016-06-16:
Just tried to update to RC from updater and receive many npm err's with unsuccessful results.

info: Starting task: 529c8b00-3414-11e6-b417-f96d7cc81cbf [updater]
shell: Stopping the engine... [hooks]
shell: Remounting root partition read-write [hooks]
shell: Resetting... [hooks]
shell: HEAD is now at 3a0482f input fixes in config as well [hooks]
shell: Checking out master... [hooks]
shell: Switched to branch 'master' [hooks]
shell: Your branch is up-to-date with 'origin/master'. [hooks]
shell: Pulling master branch and tags... [hooks]
shell: From https://github.com/FabMo/FabMo-Engine [hooks]
shell:    a243bf5..f68a9fa  master     -> origin/master [hooks]
shell:    3a0482f..f908875  rc         -> origin/rc [hooks]
shell: Updating a243bf5..f68a9fa [hooks]
shell: Fast-forward [hooks]
shell:  authentication/authentication.html                 |  202 +-- [hooks]
shell:  dashboard/apps/configuration.fma/css/style.css     |   18 + [hooks]
shell:  .../apps/configuration.fma/js/configuration.js     |    4 + [hooks]
shell:  dashboard/apps/job_manager.fma/index.html          |    5 + [hooks]
shell:  dashboard/apps/macro_manager.fma/css/style.css     |   11 +- [hooks]
shell:  dashboard/static/css/style.css                     |   30 +- [hooks]
shell:  dashboard/static/index.html                        |    7 +- [hooks]
shell:  dashboard/static/js/context.js                     |   26 +- [hooks]
shell:  dashboard/static/js/dashboard.js                   | 1813 +++++++++++--------- [hooks]
shell:  dashboard/static/js/main.js                        |   23 +- [hooks]
shell:  dashboard/static/js/routers.js                     |   21 +- [hooks]
shell:  dashboard/static/js/templates/authentication.html  |  129 ++ [hooks]
shell:  dashboard/static/js/views.js                       |   28 +- [hooks]
shell:  routes/authentication.js                           |    2 +- [hooks]
shell:  routes/index.js                                    |   22 +- [hooks]
shell:  routes/websocket.js                                |    5 +- [hooks]
shell:  16 files changed, 1380 insertions(+), 966 deletions(-) [hooks]
shell:  create mode 100644 dashboard/static/js/templates/authentication.html [hooks]
shell: Fetching release branches... [hooks]
shell: From https://github.com/FabMo/FabMo-Engine [hooks]
shell:    3a0482f..f908875  rc         -> rc [hooks]
shell: Updating to version rc... [hooks]
shell: Switched to branch 'rc' [hooks]
shell: Installing dependencies... [hooks]
shell: npm WARN package.json async@1.4.2 No README data [hooks]
shell: npm WARN package.json escape-regexp-component@1.0.2 No repository field. [hooks]
shell: npm WARN package.json svgmin@0.1.0 No description [hooks]
shell: npm WARN package.json svgmin@0.1.0 No repository field. [hooks]
shell: npm WARN package.json svgmin@0.1.0 No README data [hooks]
shell: npm ERR! makeCacheDir homeless? [hooks]
shell: npm ERR! Error: Attempt to unlock https://github.com/jlucidar/bonjour, which hasn't been locked [hooks]
shell: npm ERR!     at unlock (/usr/lib/node_modules/npm/lib/utils/locker.js:44:11) [hooks]
shell: npm ERR!     at cb (/usr/lib/node_modules/npm/lib/cache/add-remote-git.js:37:5) [hooks]
shell: npm ERR!     at /usr/lib/node_modules/npm/lib/cache/add-remote-git.js:59:20 [hooks]
shell: npm ERR!     at /usr/lib/node_modules/npm/lib/utils/locker.js:22:20 [hooks]
shell: npm ERR!     at /usr/lib/node_modules/npm/node_modules/inflight/inflight.js:22:7 [hooks]
shell: npm ERR!     at Array.forEach (native) [hooks]
shell: npm ERR!     at res (/usr/lib/node_modules/npm/node_modules/inflight/inflight.js:21:9) [hooks]
shell: npm ERR!     at f (/usr/lib/node_modules/npm/node_modules/once/once.js:16:25) [hooks]
shell: npm ERR!     at /usr/lib/node_modules/npm/lib/cache/get-stat.js:44:14 [hooks]
shell: npm ERR!     at Object.oncomplete (evalmachine.:108:15) [hooks]
shell: npm ERR! If you need help, you may report this *entire* log, [hooks]
shell: npm ERR! including the npm and node versions, at: [hooks]
shell: npm ERR!      [hooks]
shell: npm ERR! System Linux 3.10.17-yocto-standard [hooks]
shell: npm ERR! command "node" "/usr/bin/npm" "install" "--production" [hooks]
shell: npm ERR! cwd /fabmo/engine [hooks]
shell: npm ERR! node -v v0.10.40 [hooks]
shell: npm ERR! npm -v 1.4.28 [hooks]
shell: npm ERR!  [hooks]
shell: npm ERR! Additional logging details can be found in: [hooks]
shell: npm ERR!     /fabmo/engine/npm-debug.log [hooks]
shell: npm ERR! not ok code 0 [hooks]
shell: HEAD is now at f908875 Merge pull request #437 from FabMo/master [hooks]
shell: Could not install dependencies with npm [hooks]
shell: Stopping the engine...
       Remounting root partition read-write
       Resetting...
       HEAD is now at 3a0482f input fixes in config as well
       Checking out master...
       Your branch is up-to-date with 'origin/master'.
       Pulling master branch and tags...
       Updating a243bf5..f68a9fa
       Fast-forward
        authentication/authentication.html                 |  202 +--
        dashboard/apps/configuration.fma/css/style.css     |   18 +
        .../apps/configuration.fma/js/configuration.js     |    4 +
        dashboard/apps/job_manager.fma/index.html          |    5 +
        dashboard/apps/macro_manager.fma/css/style.css     |   11 +-
        dashboard/static/css/style.css                     |   30 +-
        dashboard/static/index.html                        |    7 +-
        dashboard/static/js/context.js                     |   26 +-
        dashboard/static/js/dashboard.js                   | 1813 +++++++++++---------
        dashboard/static/js/main.js                        |   23 +-
        dashboard/static/js/routers.js                     |   21 +-
        dashboard/static/js/templates/authentication.html  |  129 ++
        dashboard/static/js/views.js                       |   28 +-
        routes/authentication.js                           |    2 +-
        routes/index.js                                    |   22 +-
        routes/websocket.js                                |    5 +-
        16 files changed, 1380 insertions(+), 966 deletions(-)
        create mode 100644 dashboard/static/js/templates/authentication.html
       Fetching release branches...
       Updating to version rc...
       Installing dependencies...
       HEAD is now at f908875 Merge pull request #437 from FabMo/master
        [hooks]
info: Finishing task 529c8b00-3414-11e6-b417-f96d7cc81cbf with a state of failed [updater]
error: Did not update to rc successfully. [hooks]

jlucidar, 2016-06-16:
this is normal, you need to run your Edison on FabMo image v1.3.0 (released yesterday) to run the rc branch. I'm still playing with it right now and I will tell you want it's ready.

closed 2016-06-16 by jlucidar

jlucidar, 2016-06-17:
Ok @SW0rden , It looks to work fine on my side. This update is a bit special because we need some dependencies that are not installable with previous images. Here are the instructions for installing the release candidate :
- flash your edison with the image 1.3.0 (updated yesterday)
- connect to your tool and mount it on your network
- go to the updater and update to the release candidate (if it fails, just retry, I had it failed once and then it worked fine)
- Flash the G2 Firmware
- Install the .fmu file containing apps and macros (optionnal)
- Reboot the board (unplug and replug)
- You should be all set up !

The default credential for the authentication are : 
  - username : admin
  - password: go2fabmo  

Please change this password (be sure to remember it ! cannot be recovered, except from a factory reset) and create a new user that you will use instead of the admin account (Configuration>User).  
Please report any bug on Github, enjoy !

SW0rden, 2016-06-17:
Ok, thanks.  Did you mean image 1.5.3 ?

jlucidar, 2016-06-17:
yes ! sorry