FabMo/FabMo-Engine #446: Up/Down Arrows displayed next to running file name.

SW0rden, 2016-06-21:
A pair of scroll up/down arrows are displayed in the pop up Stop Bar next to the running file name.

Having the running file name displayed there is a bit of a redundancy, since it's displayed in the 'Now Running' box.  This might be a good place for displaying the elapsed time to go with the percentage indicator.  Items in the Stop Bar could use a bit of padding between them and the taskbar for desktop/laptop.

ryansturmer, 2016-06-21:
We don't see the redundancy because there is no guarantee that the user will remain in the job manager while the file cuts, and indeed, it is not required that they go through the job manager to start a file, in some cases.  The elapsed time will likely go inside the progress bar itself, or next to it.  We will fix the scrollbar issue, though.

wholetthedogsout, 2016-06-21:
Is this in firefox or chrome? and on PC? @SW0rden

ryansturmer, 2016-06-21:
We've seen it on PC/Chrome, and i've seen it on mac firefox as well, I think.

SW0rden, 2016-06-21:
Sorry, both Chrome and FF on desktop/laptop.

wholetthedogsout, 2016-06-27:
Should be fixed, never been able to replicate myself but so please test.

SW0rden, 2016-06-27:
Looks good now.

closed 2016-06-29 by ryansturmer