FabMo/FabMo-Engine #447: DRO decimal places in FF

SW0rden, 2016-06-21:
The decimal place values for whole numbers are not shown in FireFox.

[image]

ryansturmer, 2016-06-24:
This is a bug in firefox, unfortunately.  The issue stems from the fact that if you use the "number" input disposition for those input fields, firefox doesn't allow you to control the formatting.  We can remove the number disposition from the input, but then when you click the boxes on mobile, you get the alphabetic keypad instead of the numeric one, which is an inconvenience.  So it's a matter of choosing between proper number alignment, and proper keypad on mobile.  I guess number alignment takes priority, but it sure is a drag.

SW0rden, 2016-06-27:
What about making Fabmo a Chrome 'Standalone' app that requires Chrome to be installed, but feels more like native apps than ones tied to the browser?  Kind of like Hangouts is.  Just a thought and a way to narrow it to one browser usage platform.  Not sure what that would mean for smartphones/tablets, but guess it would be just like installing any other app for Chrome.  Could have the Fabmo Finder built in at that point maybe.

Not a big deal for those that use Chrome or FF, but there's a lot of people that know nothing other than IE or now Edge in Win 10.  Just some thoughts.

jlucidar, 2016-06-28:
@SW0rden :  It's what we did a while ago, we embedded a chrome browser with the tool minder, so you can access it directly without having another browser aside.
The reason we decided to get rid of that solution is flexibility, and user choice. In that way, the FabMo components are all independent, and you can adopt the workflow you want to use your machine.
Maybe, down the road, we will offer a packaged solution, but it also means more software maintenance to do.
When we send FabMo machines to customer, we advise them to use Chrome in priority as it is the most tested browser and the one that will provide them the best user experience.
Just for your information, all version (Mac/Linux/Windows/Android/IOS) of the tool minder already embed a chrome browser in their code.

closed 2019-10-15 by tedh-shopbot