FabMo/FabMo-Engine #450: FabMo/G2 stutter and losing steps ... continuation of Brian's issue

tedh-shopbot, 2016-06-21:
When running the new crawler with longer axis, there is a stutter at the beginning of the down ramp on most moves over about 12 inches. This happens at normal speeds and ramps, but does not happen with shorter moves (it's as if there are a couple of steps missing before the ramp starts). Sounds similar to issues we were having earlier with unusual units. Frequently the stutter will cause the tool to stall or miss steps. This is probably the problem that Gordon and Jimmy were having in Detroit on the Desktop. It is very repeatable.

tedh-shopbot, 2016-06-22:
Previous was with 1.1.1  ... I repeated this one with the current RC and updated firmware. Planner issue?

ryansturmer, 2016-06-22:
Yes, I believe these to be the same issue - probably the next big stumbling block for big tool units.  Newest RC doesn't have the adult units firmware though - do the latest master, and see where that gets you (it should squash any tr/mi/sa weirdness)

-R

SW0rden, 2016-06-23:
This sound just like what I've been experiencing from the start.  It's always/only upon deceleration.  I can get rid of it for a time by deleting the G2.json, but if I do some tweaking for jerk, velocity or feed  in config, it might come back.  I've compared the old and new G2.json files and see no difference with exception of some ABC values being different.  Current working ok with v1.1.1-113-gbff541e, but haven't had time to do much with it.
 
You can hear/see it in these two videos...

Starts at X0, loses steps when it returns to X0 the first pass and then loses again at the extent of travel in the same amount, as when it returns again, it's back to X0.   
https://youtu.be/1Crp8WIfAOY

A rectangle that should be repeated right on top of itself each time.  
https://youtu.be/7d226z60CgA

tedh-shopbot, 2016-06-23:
Having watched the videos, I'm not absolutely sure it is the same issue ... but probably. I'm thinking there is a more aggressive jerk setting in Scott's file. In the cases I have been seeing it, the jerk is very gentle, so you notice the stutter right at the start of the down ramp (well before end of move). Makes it very obvious.

I have been using FabMo a bit on a bigger tool and had not seen the problem with 20-30 inch moves, but that tool has a low unit value, so in terms of steps, it is not going as many steps as a desktop or Handibot on long moves. I'm surprised we have not noticed this on Desktops before ... perhaps we were too obsessed with small moves and arcs, thinking straight-lines were easy.

@SW0rden If this were Sb3, this would be exactly the same problem that results in our problem slow down, a slight error in the step count that determines the location of the down ramp.

ryansturmer, 2016-06-23:
There is a planner update in the works for which there is actually a release candidate.  After the release of the current fabmo rc, the plan is to start testing and integration of the new firmware, which should resolve this.

SW0rden, 2016-08-08:
Any word on this situation?  Seems like in the last few updates to RC that it has gotten worse.  I can't maintain positional accuracy at all anymore.  At at end of a move or jog, the ramp down stops abruptly short of commanded position.

SW0rden, 2016-08-10:
I can cut other shapes/designs, but when cutting a rectangle as in the above attached video, it stops short of the corners and will generally do it by the same amount on the opposite corner which pretty much resets location to what it started at.  Running 50,50,100 on jerk settings.  This has been present since last November, time to get it quashed!  What do I need to do on my end or is it totally on the G2 end?

For reference, see attached pic showing the mis-cut rectangle.  This is a v-carve rectangle, not a on the line rectangle.  Could the v-carve code have something to do with it (meaning the reading of many lines of small moves compared to one line of a move consisting of the same distance or same rectangle)?

[image]

tedh-shopbot, 2016-08-10:
Hey Scott, Can you send me this file to putter with. Also you config values (tool-ip/config). Thanks, I'm still struggling with issues like these.

Also, I'm thinking that you are running 1.2.0 and have updated the firmware. Correct?

=Ted

ryansturmer, 2017-02-15:
This is resolved, AFAIK in the latest firmware.

closed 2017-02-15 by ryansturmer