FabMo/FabMo-Engine #460: Preview of .nc files no longer works

SW0rden, 2016-06-27:
Pretty sure it used to work, but it's been a while since I've used it.

ryansturmer, 2016-06-27:
Can you be more specific?  What happens when you attempt to preview a file?

SW0rden, 2016-06-27:
Hangs there forever.

[image]

ryansturmer, 2016-06-27:
Please post the file, and the version of software you're running Scott, we don't observe the error with any of our usual test files on the latest RC.

SW0rden, 2016-06-27:
v1.1.1-129-g24feab6

Here's the original file...
[Peg Storage Cover.nc.txt](https://github.com/FabMo/FabMo-Engine/files/335656/Peg.Storage.Cover.nc.txt)

Files edited to narrow it down...
[Preview Fails.nc.txt](https://github.com/FabMo/FabMo-Engine/files/335729/Preview.Fails.nc.txt)
[Preview Works.nc.txt](https://github.com/FabMo/FabMo-Engine/files/335730/Preview.Works.nc.txt)

It appears that it fails where the code/cut starts a spiral down toolpath from Aspire.  I've commented the place where this segment starts in the 'Fails' file.  Too many small segments of positional changes from line to line?

Alex-Canales, 2016-06-28:
Hello SWo0rden, I am the developper maintaining the Viewer app. I wish to thank you for pointing out those issues.

The Viewer app includes a library transforming the G-Code into geometrical shapes. The problem you had came from this library. When checking if the code is mathematically possible or no, an error was not handled correctly which caused the issue. Also, the precision for equality testing was too high.

Also, the file ``Preview Fails.nc.txt``

    'G3 X-2.0762 Y-1.9008 I0.0731 J-0.3510 
    'G3 X-1.7908 Y-2.2518 I0.3586 J0.0000

What ``'`` is supposed to mean? This character causes an issue in the lexical parser made by ryansturmer.

    SyntaxError: Expected "A", "B", "C", "D", "F", "G", "H", "I", "J", "K", "L", "M", "N", "P", "Q", "R", "S", "T", "X", "Y", "Z" or end of input but "'" found.

jlucidar, 2016-06-28:
@Alex-Canales in openSBP,  `'`  mark a line as a comment.
There is a good online documentation about sbp language at www.opensbp.com

Alex-Canales, 2016-06-28:
@jlucidar thank you for the information. I thought the file was pure G-Code.

jlucidar, 2016-06-28:
Oops, yes the file is a plain gcode file, but uses openSBP comment style... is it intended @SW0rden ?

ryansturmer, 2016-06-28:
Yeah, ' does not connote a comment in G-Code, comments are done like this:

```
(This is a comment)
```

I have observed that some shopbot post processors mix the SBP comment style and the g-code one.  We should fall back on being permissive here, although we shouldn't go beyond what G2 tolerates.  I think as it is, G2 will simply ignore a comment line of either type, and so our previewer should too.

SW0rden, 2016-06-28:
Hi Alex,

Sorry, I put those in there for testing purposes intending to comment out lines that I wanted to skip and just out of habit, used the apostrophe when I should have used one of the standard gcode comment characters.  I use opensbp most of the time and just forgot that I needed a different character.  The original file doesn't have them in it and if you chop off the code below line 831, it previews fine (that's where the spiral ramp starts).

Part of this habit comes from SB3 being able to mix a bit of both formats.  As seen in the original file, I use ; for commenting and this choice came from using the TinyG post processor from Vectric as a base example for creating my own FabMo_Gcode_Inch.pp.

SW0rden, 2016-06-28:
Just a note, the same file in opensbp, while it takes a little time to load in the previewer, it does preview fine.  Rename to just .sbp 

[Covers.sbp.txt](https://github.com/FabMo/FabMo-Engine/files/337757/Covers.sbp.txt)

Alex-Canales, 2016-06-28:
No problem @SW0rden. Without the lines with the apostrophe, the code runs fine.

I will try to update in the engine the code during the week.

ryansturmer, 2016-06-28:
I think there's no reason not to be permissive of ' : ; " etc as comment characters in the g-code parser, given that I believe G2 just skips right over them.  The previewer should ideally never fail a file that would pass on the real hardware, and it should never pass a file that would be a hardware fail.

SW0rden, 2016-06-28:
On the topic of being 'permissive', one thing I'd like to see would be the option of passing a message to the screen/user with nc like can be done with opensbp.  For example, Pause "message here" and/or msgbox.

ryansturmer, 2016-06-29:
We're not going to do that in the near term, and i think we've resolved the comment issue, so I'm closing this ticket.

commit 4ea07dbb (2016-06-29): Update the Viewer. Fix #460.
closed 2016-06-29 by ryansturmer