FabMo/FabMo-Engine #469: Stuck Z in keypad move seems very likely at Handiland today ...

tedh-shopbot, 2016-06-29:
This is in android mobile (on the wimpy network here) ... tapping a few times seems invariably to locks it; tap releases. In this case, it is worse than not having the discrete move ... it is happening unpredictably with both short and long taps.

ryansturmer, 2016-06-29:
When you say stuck Z - what do you mean exactly?  Is the Z running away?  Or is it just not being responsive?  What is the state of the button?

tedh-shopbot, 2016-06-29:
Running away after removing touch. Stays activated. Tapping again de-activates and stops it. Thus stuck means stuck-on and moving; and, thus feeling out-of-control. I'm less concerned about being un-responsive about starting. Of course, this gets back to my original concern with keypads and my original enthusiasm for requiring active rotation of the spinner in order to make use of a repeating stimulus, rather than relying on sensing the touch/button going off.

wlyoung, 2016-06-30:
Don't know if this is related, but I've had some weird keypad movements today. Occasional 2-axis moves when hitting an arrow, with the tool moving in both Y and Z when clicking the Y arrow, or moving up when clicking the Z-down arrow.

wlyoung, 2016-06-30:
Oh yeah, this is on a PC with release candidate v1.1.1-134-g5636650

tedh-shopbot, 2016-11-14:
Stuck Keypad moves continue to be a problem for me and others.

Yesterday I noticed a couple of things:
- It is usually happening for me, the first or second time I use the Keypad after powering up.
- It seems to be happening on PC, when using mouse to activate key ... and particularly when, when using the touch pad with one finger on the pad and the other on the left button. Something about the timing of the change on the two keys, perhaps (in all cases, the keypad stays highlighted after fingers are removed. Touching it again stops the move.
- My computer has sometimes misled us because it has a touchscreen ... which I never use.

closed 2018-12-06 by tedh-shopbot