FabMo/FabMo-Engine #473: Stuck when paused if previous job ends when paused

Alex-Canales, 2016-07-04:
If you pause a job and at the same moment the job finishes, then when the next job is paused, the buttons "resume" and "quit" will not appear and you will get stuck.

How to reproduce the bug:

Step 1: use this code sample:

    G0 X1 Y1 Z1
    G0 X0 Y0 Z0

Step 2: run the job. When the job is nearly done (x, y, z nearly equal to 0), press the stop button. The job should be over with the stop button activated (and the buttons "resume" and "quit" displayed). The buttons "resume" and "quit" disappear automatically.

Step 3: run a new job. Press the stop button. The buttons "resume" and "quit" will not appear and the job manager will be stuck with this job. To stop the problem, you will have to kill the engine and run it again.

Here is the result at step 3.

[image]

Alex-Canales, 2016-07-05:
I manage to get stuck by putting two jobs (always the same as in the example) in the queue, launching one, pausing, resuming. When the first one is over, the second one is display as on the previous .screenshot

jlucidar, 2016-07-05:
OK, I'm trying to reproduce it. just to make sure : Can you copy and paste the log from the engine here ?

Alex-Canales, 2016-07-06:
Here what is print in the terminal. Two jobs are pending in the queue. The logs are from the moment I run the first job.

    debug: POST /jobs/queue/run [engine]
    info: Running the next job in the queue [routes]
    info: Arming the machine for nextJob [machine]
    info: Firing automatically since authorization is disabled. [machine]
    info: Machine is deauthorized. [machine]
    info: Running next job [machine]
    info: printing[object Object] [db]
    info: 72 [db]
    info: [object Object] [machine]
    info: Running job {"file_id":5,"name":"Less dumb short code","description":"A G-Code job from the editor","created_at":1467799242213,"started_at":1467799259089,"finished_at":null,"state":"running","order":null,"_id":72} [machine]
    info: Running file /opt/fabmo/files/b295c760-42ab-11e6-906b-f9cd58d69b79.nc [machine]
    info: Got a machine state change: idle [machine]
    info: Connected G-Code Runtime [gcode]
    debug: Saving config to /opt/fabmo/config/instance.json [config]
    debug: GET /jobs/queue [engine]
    debug: GET /jobs/queue [engine]
    debug: fsync()ed /opt/fabmo/config/instance.json [config]
    debug: GET /jobs/history?start=0&count=10 [engine]
    info: Got a machine state change: running [machine]
    info: Got a machine state change: paused [machine]
    debug: Saving config to /opt/fabmo/config/instance.json [config]
    debug: fsync()ed /opt/fabmo/config/instance.json [config]
    info: Arming the machine for resume [machine]
    info: Firing automatically since authorization is disabled. [machine]
    info: Machine is deauthorized. [machine]
    info: Got a machine state change: running [machine]
    info: 72 [db]
    info: Setting driver units to INCH [g2]
    debug: GET /jobs/queue [engine]
    debug: GET /jobs/history?start=0&count=10 [engine]
    info: Setting driver units to INCH [g2]
    info: Got a machine state change: idle [machine]
    debug: Saving config to /opt/fabmo/config/instance.json [config]
    debug: fsync()ed /opt/fabmo/config/instance.json [config]
    info: Got a machine state change: idle [machine]
    debug: Saving config to /opt/fabmo/config/instance.json [config]
    debug: fsync()ed /opt/fabmo/config/instance.json [config]
    debug: POST /jobs/queue/run [engine]
    info: Running the next job in the queue [routes]
    info: Arming the machine for nextJob [machine]
    info: Firing automatically since authorization is disabled. [machine]
    info: Machine is deauthorized. [machine]
    info: Running next job [machine]
    info: printing[object Object] [db]
    info: 73 [db]
    info: [object Object] [machine]
    info: Running job {"file_id":5,"name":"Less dumb short code","description":"A G-Code job from the editor","created_at":1467799245346,"started_at":1467799264747,"finished_at":null,"state":"running","order":null,"_id":73} [machine]
    info: Running file /opt/fabmo/files/b295c760-42ab-11e6-906b-f9cd58d69b79.nc [machine]
    debug: GET /jobs/queue [engine]
    debug: GET /jobs/queue [engine]
    debug: GET /jobs/history?start=0&count=10 [engine]

Does the engine generate log files that would me to copy?

ryansturmer, 2016-07-06:
Alex, can you perform the same test, but turn your log level up to "g2" ?

Alex-Canales, 2016-07-08:
Here is the logs. As you can see, there is a "no pending job" error. Two jobs are pending in the queue. The logs are from the moment I run the first job.

    debug: POST /jobs/queue/run [engine]
    info: Running the next job in the queue [routes]
    info: Arming the machine for nextJob [machine]
    info: Firing automatically since authorization is disabled. [machine]
    info: Machine is deauthorized. [machine]
    debug: Firing a nextJob [machine]
    info: Running next job [machine]
    info: printing[object Object] [db]
    info: 75 [db]
    info: [object Object] [machine]
    info: Running job {"file_id":5,"name":"Less dumb short code","description":"A G-Code job from the editor","created_at":1467999715411,"started_at":1467999846267,"finished_at":null,"state":"running","order":null,"_id":75} [machine]
    info: Running file /opt/fabmo/files/b295c760-42ab-11e6-906b-f9cd58d69b79.nc [machine]
    info: Got a machine state change: idle [machine]
    g2: --C-1467999846278----> {"mpo":null} [g2]
    info: Connected G-Code Runtime [gcode]
    g2: <-C--1467999846281---- {"r":{"mpo":{"x":0.000,"y":0.000,"z":0.000,"a":0.000,"b":0.000,"c":0.000}},"f":[1,0,13]} [g2]
    debug: Saving config to /opt/fabmo/config/instance.json [config]
    debug: GET /jobs/queue [engine]
    debug: GET /jobs/queue [engine]
    debug: GET /jobs/history?start=0&count=10 [engine]
    g2: --G-1467999846416----> G90
    N1G0 X1 Y1 Z1
    N2G0 X2 Y2 Z2
    N3G0 X0 Y0 Z0
    M30

    M30 [g2]
    debug: fsync()ed /opt/fabmo/config/instance.json [config]
    g2: <-C--1467999846537---- {"sr":{"posx":0.004,"posy":0.004,"posz":0.004,"vel":24.53,"stat":5,"line":1}} [g2]
    info: Got a machine state change: running [machine]
    g2: <-C--1467999846788---- {"sr":{"posx":0.632,"posy":0.632,"posz":0.632,"vel":415.69}} [g2]
    g2: <-C--1467999847039---- {"sr":{"posx":1.626,"posy":1.626,"posz":1.626,"vel":387.88,"line":2}} [g2]
    g2: <-C--1467999847290---- {"sr":{"posx":2.000,"posy":2.000,"posz":2.000,"vel":8.67,"line":3}} [g2]
    g2: <-C--1467999847540---- {"sr":{"posx":1.592,"posy":1.592,"posz":1.592,"vel":395.41}} [g2]
    g2: <-C--1467999847790---- {"sr":{"posx":0.596,"posy":0.596,"posz":0.596,"vel":415.62}} [g2]
    debug: Sending a feedhold [g2]
    g2: --C-1467999847990----> !
     [g2]
    debug: Drained. [g2]
    g2: <-C--1467999848042---- {"sr":{"posx":0.003,"posy":0.003,"posz":0.003,"vel":18.85,"stat":6,"hold":4}} [g2]
    info: Got a machine state change: paused [machine]
    g2: --C-1467999848043----> {"mpo":null} [g2]
    g2: <-C--1467999848054---- {"r":{"mpo":{"x":0.002,"y":0.002,"z":0.002,"a":0.000,"b":0.000,"c":0.000}},"f":[1,0,13]} [g2]
    debug: Saving config to /opt/fabmo/config/instance.json [config]
    g2: <-C--1467999848096---- {"er":{"fb":89.02,"st":94,"msg":"Planner assertion failure - zero length line in calculate trapezoid"}} [g2]
    g2: <-C--1467999848096---- {"er":{"fb":89.02,"st":94,"msg":"Planner assertion failure - zero length move in exec_aline"}} [g2]
    debug: fsync()ed /opt/fabmo/config/instance.json [config]
    g2: <-C--1467999848288---- {"sr":{"posx":0.000,"posy":0.000,"posz":0.000,"vel":0.00,"hold":7}} [g2]
    info: Arming the machine for resume [machine]
    info: Firing automatically since authorization is disabled. [machine]
    info: Machine is deauthorized. [machine]
    debug: Firing a resume [machine]
    g2: --C-1467999850405----> ~ [g2]
    g2: <-C--1467999850407---- {"sr":{"stat":4,"hold":0,"line":0}} [g2]
    info: 75 [db]
    info: Setting driver units to INCH [g2]
    g2: --C-1467999850411----> {"gun":0} [g2]
    g2: <-C--1467999850413---- {"r":{"gun":0},"f":[1,0,10]} [g2]
    g2: --G-1467999850413----> G20
    M30 [g2]
    debug: GET /jobs/queue [engine]
    debug: GET /jobs/history?start=0&count=10 [engine]
    g2: <-C--1467999850663---- {"sr":{"stat":4,"unit":0}} [g2]
    info: Setting driver units to INCH [g2]
    g2: --C-1467999850664----> {"gun":0} [g2]
    info: Got a machine state change: idle [machine]
    g2: --C-1467999850665----> {"mpo":null} [g2]
    g2: <-C--1467999850666---- {"r":{"gun":0},"f":[1,0,10]} [g2]
[... 132 lines cut ...]
    g2: <-C--1467999872337---- {"sr":{"posx":0.920,"posy":0.920,"posz":0.920,"stat":4}} [g2]
    g2: <-C--1467999872589---- {"sr":{"posx":0.880,"posy":0.880,"posz":0.880,"stat":4}} [g2]
    g2: <-C--1467999872840---- {"sr":{"posx":0.841,"posy":0.841,"posz":0.841,"stat":4}} [g2]
    g2: <-C--1467999873092---- {"sr":{"posx":0.801,"posy":0.801,"posz":0.801,"stat":4}} [g2]
    g2: <-C--1467999873343---- {"sr":{"posx":0.762,"posy":0.762,"posz":0.762,"stat":4}} [g2]
    g2: <-C--1467999873596---- {"sr":{"posx":0.722,"posy":0.722,"posz":0.722,"stat":4}} [g2]
    g2: <-C--1467999873846---- {"sr":{"posx":0.683,"posy":0.683,"posz":0.683,"stat":4}} [g2]
    g2: <-C--1467999874097---- {"sr":{"posx":0.643,"posy":0.643,"posz":0.643,"stat":4}} [g2]
    g2: <-C--1467999874348---- {"sr":{"posx":0.604,"posy":0.604,"posz":0.604,"stat":4}} [g2]
    g2: <-C--1467999874598---- {"sr":{"posx":0.565,"posy":0.564,"posz":0.564,"stat":4}} [g2]
    g2: <-C--1467999874849---- {"sr":{"posx":0.525,"posy":0.525,"posz":0.525,"stat":4}} [g2]
    g2: <-C--1467999875100---- {"sr":{"posx":0.486,"posy":0.486,"posz":0.486,"stat":4}} [g2]
    g2: <-C--1467999875351---- {"sr":{"posx":0.446,"posy":0.446,"posz":0.446,"stat":4}} [g2]
    g2: <-C--1467999875603---- {"sr":{"posx":0.406,"posy":0.406,"posz":0.406,"stat":4}} [g2]
[... cut ...]
    g2: <-C--1467999877108---- {"sr":{"posx":0.170,"posy":0.170,"posz":0.170,"stat":4}} [g2]
    g2: <-C--1467999877359---- {"sr":{"posx":0.130,"posy":0.130,"posz":0.130,"stat":4}} [g2]
    g2: <-C--1467999877610---- {"sr":{"posx":0.091,"posy":0.091,"posz":0.091,"stat":4}} [g2]
    g2: <-C--1467999877861---- {"sr":{"posx":0.051,"posy":0.051,"posz":0.051,"stat":4}} [g2]
    g2: <-C--1467999878114---- {"sr":{"posx":0.012,"posy":0.012,"posz":0.012,"vel":14.44,"stat":4}} [g2]
    g2: <-C--1467999878361---- {"sr":{"posx":0.000,"posy":0.000,"posz":0.000,"vel":0.00,"stat":4,"line":0}} [g2]

jlucidar, 2017-01-24:
It seems to be fixed, I did follow your explanation and was not able to reproduce the bug.
please reopen the issue if it happens again.

closed 2017-01-24 by jlucidar