FabMo/FabMo-Engine #488: Machine parameters in config not updating

SW0rden, 2016-07-22:
Version v1.2.0

I was attempting to change the xy manual speed settings and noticed that changes are not being applied right away or at all.  If I go back to the job manager, the Job Manager graphics may or may not show or the gear shows and stays.  A couple of times, upon going back to config/machine, the value fields were all empty and after a little, they showed up.  Speed settings don't apply to motion unless you leave config and return or do something else first.  Makes me wonder what other settings may not take immediately.  This is on Chrome and happened on a PC and laptop.

This also seems to corrupt jerk settings or something too, as I occasionally get the abrupt stop/loss of step situation that has been mentioned previously by myself and Ted. 

Kind of strange and I'll keep trying to see if there's something more concrete/repeatable that I can post on it, but maybe someone can give it a try to see if it's repeatable there.

SW0rden, 2016-08-08:
I've had this happen a number of times now.  I go into config to adjust something and values aren't updated and If I try to go to job manager, it hangs at the gear.  If I then try to go back to config, all values are blank and may or may not eventually appear.   Once values appear, some that were attempted to be changed haven't been updated.

ryansturmer, 2016-08-08:
Scott, 

Can you say specifically what setting you're adjusting (and what you're changing the value to) that gives you the behavior?

SW0rden, 2016-08-08:
Initially it was just the machine manual control speed, but later on, I dumped the G2 and instance files again and while filling in all parameters for general, axis, inputs, and channels, I started seeing that values weren't updating.   Upon leaving config and coming back in, even the untouched values of machine name and unit system etc. were wrong/gone and set to metric.  At some point, they finally refresh, but some values like unit values that were set came back as default values (12000 back to 4000).  Manual control speed values don't always actually change the speed until leaving config and coming back into it.  Leaving to job manager is the path that hangs at the gear, but if I go to home, it doesn't seem to hang.

ryansturmer, 2016-08-30:
This is closed by #506

closed 2016-08-30 by ryansturmer